carla-simulator/carla #3426: Lightning problem in case of a new map in the latest build 0.9.10

mercurylaci, 2020-10-11:
I'm using Carla 0.9.10, Unreal Engine 4.23.3 on Ubuntu 18.04.
On the previous build 0.9.9 I don't have this issue, when I import a map (which was created by roadrunner), in the Unreal Editor there is proper lightning, but when I hit the play button (or create a package), there is complete darkness.
[image]
[image]

And I have this error:
[image]

I tried to change the weather in Carla, add more lights to the map, removed the default one and add new one, none of them helped. I've imported these maps in previous Carla versions (0.9.8-0.9.9) and worked perfectly.

Does anyone knows the solution for this problem? 
Thanks in advance!

Madecu, 2020-10-13:
Hi @mercurylaci, in the 0.9.10 release we had changed the lighting in the scene. We merged all the lighting assets inside a Blueprint.
To solve the issue you just need to delete the Light Source (Directional Light), ExponentialHeightFog, Skylight, and SkySphere from your scene and drag BP_Sky into it.

[image]

mercurylaci, 2020-10-13:
Thank you! It's working now.

closed 2020-10-13 by mercurylaci

EchoYJJ, 2020-12-22:
Hi! @Madecu ,I do as you recommand.
The scene shows up when I am high. As I getting lower to approach the road, it get darker so that I can't view it.
Do you know how to solve it? Thanks in advance!
[image]

seuwcs, 2021-04-14:
> Hi @mercurylaci, in the 0.9.10 release we had changed the lighting in the scene. We merged all the lighting assets inside a Blueprint.
> To solve the issue you just need to delete the Light Source [...]

Hello, I have a small question that is which folder should the BP_sky dragged into? @Madecu  @mercurylaci

seuwcs, 2021-04-14:
Hello Friends. After deleting the Light Source (Directional Light), ExponentialHeightFog, Skylight, and SkySphere from the scene and drag BP_Sky into the scene. I clicked Play, and run the scrip "manual_control.py" in  the folder "carla/PythonAPI/examples", a mistake occurrd that is :

WARNING: cannot parse georeference: ''. Using default values. 
No recommended values for 'speed' attribute
There are no spawn points available in your map/town.
Please add some Vehicle Spawn Point to your UE4 scene.
[image]

I'd like to know if it is useful to add some spawn points in my scene to solve my problem, as I have added some spawn points in the scene, but the questions still remains.
Does anyone knows the solution for this problem?
Thanks in advance!

@Madecu @mercurylaci

xuhan417, 2021-05-14:
@seuwcs 
 This warning means your xodr file is missing georeference header. 
Also, your map does not have spawpoints. You can use "opendrive export preview tool" in roadrunner to check your export and look for any warnings. And you should use the "slip road tool" in the roadrunner to connect the intersections. Once you connect the roads properly the waypoint will be generated automatically.