carla-simulator/carla #3428: Error when running make PythonAPI on Linux18.04

GitJasonseven, 2020-10-12:
CARLA VERSION:0.9.8
OS: ubuntu118.04
boost_1_72_0.tar.gz. 100%[====================>] 120.72M  2.00MB/s    用时 77s   s

2020-10-12 15:09:02 (1.57 MB/s) - 已保存 “boost_1_72_0.tar.gz.3” [126580835/126580835])

Setup.sh: Extracting boost for Python 3.

gzip: stdin: unexpected end of file
tar: 归档文件中异常的 EOF
tar: 归档文件中异常的 EOF
tar: Error is not recoverable: exiting now
Util/BuildTools/Linux.mk:143: recipe for target 'setup' failed
make: *** [setup] Error 2

qhaas, 2020-10-13:
Rebuilt CARLA 0.9.8 from source (fresh-checkout) this morning on Ubuntu 18.04 without that error, check to see if you can download  / extract that boost tarball manually.

closed 2020-12-25 by stale[bot]

pengyuan-zhou, 2021-01-27:
@GitJasonseven 
you need to add the file name to be saved in Setup.sh
As the output shows, the system automatically added a ".3"  at the end of the file extension, that's what caused the error.
Can use wget -O to determine the file name to avoid this error.

seuwcs, 2021-03-26:
Dear friends, I also met a problem when Making PythonAPI, could you give me some suggestions?
the problem is list as follows:

wcs@wcs:~/carla$ make PythonAPI
Setup.sh: Retrieving libc++.
正克隆到 &apos;llvm-8.0-source&apos;...
remote: Enumerating objects: 36751, done.
remote: Counting objects: 100% (36751/36751), done.
remote: Compressing objects: 100% (32726/32726), done.
remote: Total 36751 (delta 5699), reused 14748 (delta 3353), pack-reused 0
接收对象中: 100% (36751/36751), 50.28 MiB | 286.00 KiB/s, 完成.
处理 delta 中: 100% (5699/5699), 完成.
正在检出文件: 100% (35358/35358), 完成.
正克隆到 &apos;llvm-8.0-source/projects/libcxx&apos;...
fatal: unable to access &apos;https://github.com/llvm-mirror/libcxx.git/&apos;: Failed to connect to github.com port 443: 连接超时
Util/BuildTools/Linux.mk:143: recipe for target &apos;setup&apos; failed
make: *** [setup] Error 128

Thanks a lot! @pengyuan-zhou  @qhaas @GitJasonseven

YimingChen-Real, 2021-11-01:
Just delete all boost_1_72_0.tar.gz files in path /carla/build and try ”make pythonapi“ again
Because of your bad internet there are many wrong files

CHILLed144, 2025-05-23:
> Just delete all boost_1_72_0.tar.gz files in path /carla/build and try ”make pythonapi“ again Because of your bad internet there are many wrong files

As of today, the reason is not "bad internet", but a wrong URL.

I had problems extracting boost for python3, even though i am building CARLA 0.9.13, and i solved them by following the instructions of a comment in another issue

https://github.com/carla-simulator/carla/issues/8675#issuecomment-2664908362

hope this helps