carla-simulator/carla #3430: RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

akshatamk, 2020-10-12:
Hello,
   I'm getting this error while running the python API in CARLA.
   I tried changing time-out duration as well as port number, but still the same issue. Please help

tinmodeHuang, 2020-10-12:
just now I have ran into same error and had no idea how to solve it as well, however, I can perform spawn_npc.py successfully after changing into newest edition

jackbart94, 2020-10-13:
Hi,

Could you tell us more about this, for example the steps you are taking to execute CARLA? Are you running it in Unreal Engine, or in the package version?

akshatamk, 2020-10-13:
Thanks for the response.
yes i'm running CARLA in unreal engine 4.24x.The simulator is opening but unable to run the example APIs.

jackbart94, 2020-10-13:
Did you press the Play button (or alternatively Alt+P) after launching Unreal Engine, before running the scripts? Because I've only seen this error when you forget to do so.

akshatamk, 2020-10-14:
Currently I'm trying to set up CARLA on azure Windows VM with (NC6s_V3-Tesla K80), does this environment support Unreal engine(as CARLA runs on unreal engine)? or should be configured on GeForce graphics ?

nagasanjay, 2020-10-14:
Did any one of you deleted the contents of `config` folder to load CarlaEditor fast ?

Because I did that and ran into similar issue. I solved it by downloading the config files again and putting it in config folder.

germanros1987, 2021-01-26:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2021-01-26 by germanros1987

omkarshindetof, 2021-01-29:
Facing the same issue.

omkarshinde22, 2021-01-29:
facing the same issue

by-seong-me, 2021-02-01:
Before run python script, you should run server. In my case, something like this, 
$ ./CarlaUE4.sh /Game/Maps/Town01 -carla-server -benchmark -fps=15 -windowed -ResX=800 -ResY=600

And then, you can run python script in python example directory
$ python3 manual_control.py

narayanaraju10, 2021-02-12:
Hello guys, recently while working with carla in python I used to get this error "RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000". 
This problem can be due to carla is not loading in 2000ms. in my case, i moved to higher GPU and RAM pc now its working fine and carla is loading with in 2000ms.

if you are stuck with same pc, kindly increase to pc to performance mode.

Best,
Narayana

omkarshinde22, 2021-02-12:
Resolved this issue.
Way to solve this issue.
If you are using windows then please check your directx. Presently microsoft took off directx from its official website. But you can download it from third party.
Else you can increase the ram and rerun the Carla simulator.

narayanaraju10, 2021-02-13:
change the time in   "client.set_timeout(2.0)" in the code to "client.set_timeout(200.0)"  # 200 depicts the loading time.

Best,
Narayana

pinnintipraneethkumar, 2021-04-01:
> Hello guys, recently while working with carla in python I used to get this error "RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to [...]

Hii
May i know the what is your pc  specifications. 
Thanks.

josephzhaoballer, 2021-05-02:
> Hello guys, recently while working with carla in python I used to get this error "RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to [...]

I changed the timeout to 5 seconds and it worked.

NikhilKalghatgi, 2021-05-03:
> I changed the timeout to 5 seconds and it worked.

Can you please tell in which file you increased the timeout?

josephzhaoballer, 2021-05-03:
client.set_timeout() look for this in which ever file you are running

manju838, 2021-05-16:
Thanks @narayanaraju10 ,it worked for timeout as 5.0.Mine is a 2GB Graphics card and 2.0 time didn't work.Better the graphics,lesser timeout needed

DonadiHarshith, 2021-09-25:
> Hi,
> 
> Could you tell us more about this, for example the steps you are taking to execute CARLA? Are you running it in Unreal Engine, or in the package version?

I am using package version can you help me

shsrinivgit, 2021-10-05:
> Did you press the Play button (or alternatively Alt+P) after launching Unreal Engine, before running the scripts? Because I've only seen this error when you forget to do so.

THANKS. I JUST WASTED TIME REBUILDING THE CARLA

e-Dylan, 2021-11-29:
tl;dr:
If it was open and working before you closed it, end the background process in task manager before you re-run the UE4 exe.

I just got this issue when I opened the CarlaUE4.exe unreal executable for carla, closed it, and tried to re-run it. I didn't realize the first one I had closed was still running an open process only visible in task manager. Once I found it in task manager (takes like 4gb memory but no cpu, not hard to find) and ended the task, api commands worked again after re-running the executable.

gaooogao, 2021-12-08:
problem-solving methods:
I started the task manager and ended all related software.
then i restarted the emulator . Then ran the python script. 
this problem has been sloved.

if you have this problem, maybe you can try the methods. i hope it can work.

The process :
0.  when i use the version carla-0.9.13. 
1. I get the folder after I unzip the package
2. i executed CarlaUE4.exe. Then the it work.I can see Carla's map simulation interface.
3. Then I ran the Python script. i can control the car ,and change different cars.
4. Before I want to exit the Python script, however, I have already exited the emulation screen .
5. so I restarted the emulator and closed the Python script.
6. This error occurs when I start emulation again and run the Python script  .
7. Finally, I solved the problem through the above methods

RocaPiedra, 2022-02-22:
> tl;dr: If it was open and working before you closed it, end the background process in task manager before you re-run the UE4 exe.
>
> I just got this issue when I opened the CarlaUE4.exe unreal [...]

Hey, if you are using Windows and you randomly encounter this issue (for me it started happening right after performing some tests successfully, after restart it worked fine), this is the way to solve it!
Thanks for the info, Dylan

Pratham544, 2022-04-27:
> tl;dr: If it was open and working before you closed it, end the background process in task manager before you re-run the UE4 exe.
>
> I just got this issue when I opened the CarlaUE4.exe unreal [...]

> tl;dr: If it was open and working before you closed it, end the background process in task manager before you re-run the UE4 exe.
>
> I just got this issue when I opened the CarlaUE4.exe unreal [...]

Thanks a lot, I tried increasing the timeout but it didn't work. Then i did this and it started working.

RocaPiedra, 2022-04-27:
This issue happens in both Linux and Windows. In my experience, it mostly happens when you try to close the Carla Server from the application/script used. This is very annoying as I'm trying to create sort of an app that is self-managed (including Carla Server). I launch the server through a Python subprocess and end the subprocess when execution is finished but here is when the background process is an issue.
Does anybody know how to end the Carla Server cleanly?

winterwindwang, 2022-07-20:
Same issue solved by increasing the timeout(2->10) of client.set_timeout(timeout)

Kenneth71-P, 2022-07-28:
When I had the same problem, I entered task manager and ended CarlaUE4 app. Then it solved the problem.

amirBgazand, 2024-08-29:
I had the same issue, and every time I get to this problem I just change the port to 3000, and amazingly, it works well.
just run the carlaUE4 by
CarlaUE4.exe -carla-rpc-port=3000
and in your code change the port to 3000 
client = carla.Client('127.0.0.1',3000)

chiru42, 2024-10-22:
> tl;dr: If it was open and working before you closed it, end the background process in task manager before you re-run the UE4 exe.
>
> I just got this issue when I opened the CarlaUE4.exe unreal [...]

This was happening to me. Thanks for mentioning about the hidden Background process. I end it from taskmanager and a fresh launch solved my issue.