carla-simulator/carla #3431: carla docker runs wrong

yangyueren, 2020-10-12:
docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.10.1
sh: 1: xdg-user-dir: not found

yangyueren, 2020-10-12:
When I try the dockerfile, it goes wrong too.
https://github.com/carla-simulator/carla/tree/master/Util/Docker

Step 8/9 : RUN  ./GenerateProjectFiles.sh &&   make
 ---> Running in 63c6a77889e8
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
        updated UE4Printers.py
        no ~/.gdbinit file found - creating a new one.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
ERROR: Unable to instantiate instance of 'BenchmarkToolTarget' object type from compiled assembly 'UE4ProgramRules'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> UnrealBuildTool.BuildException: GetBuildPlatform: No BuildPlatform found for Linux
         at UnrealBuildTool.UEBuildPlatform.GetBuildPlatform (UnrealBuildTool.UnrealTargetPlatform InPlatform, System.Boolean bInAllowFailure) [0x0003e] in <0f62df6941b7421da985bf66bc90ab8a>:0
         at UnrealBuildTool.TargetRules..ctor (UnrealBuildTool.TargetInfo Target) [0x002f2] in <0f62df6941b7421da985bf66bc90ab8a>:0
         at BenchmarkToolTarget..ctor (UnrealBuildTool.TargetInfo Target) [0x00000] in <a0632d70062046989cbf014c520bc9e6>:0
         at (wrapper managed-to-native) System.Reflection.MonoCMethod.InternalInvoke(System.Reflection.MonoCMethod,object,object[],System.Exception&)
         at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <98fac219bd4e453693d76fda7bd96ab0>:0
          --- End of inner exception stack trace ---
         at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00017] in <98fac219bd4e453693d76fda7bd96ab0>:0
         at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0007a] in <98fac219bd4e453693d76fda7bd96ab0>:0
         at System.Reflection.MonoCMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0002f] in <98fac219bd4e453693d76fda7bd96ab0>:0
         at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0
         at UnrealBuildTool.RulesAssembly.CreateTargetRulesInstance (System.String TypeName, UnrealBuildTool.TargetInfo TargetInfo) [0x00093] in <0f62df6941b7421da985bf66bc90ab8a>:0
The command '/bin/sh -c ./GenerateProjectFiles.sh &&   make' returned a non-zero code: 5

gismo07, 2020-10-13:
> docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.10.1
> sh: 1: xdg-user-dir: not found

I had the same issue. 
Starting the container with a shared X context works:

```bash
docker run \
  -e SDL_VIDEODRIVER=x11 \
  -e DISPLAY=:99 \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -p 2000-2002:2000-2002 \
  -it \
  --gpus all \
  carlasim/carla:0.9.10.1 ./CarlaUE4.sh -opengl
```

btw you don't have to pass `--runtime=nvidia`.

zchrissirhcz, 2020-10-14:
> > docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.10.1
> > sh: 1: xdg-user-dir: not found
> 
> I had the same issue.
> Starting the container with a shared X context works:
> 
> ```shell
> docker run \
>   -e SDL_VIDEODRIVER=x11 \
>   -e DISPLAY=$DISPLAY\
>   -v /tmp/.X11-unix:/tmp/.X11-unix \
>   -p 2000-2002:2000-2002 \
>   -it \
>   --gpus all \
>   carlasim/carla:0.9.10.1 ./CarlaUE4.sh -opengl
> ```
> 
> btw you don't have to pass `--runtime=nvidia`.

I execute the pasted command but got error:
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
No protocol specified
No protocol specified
No protocol specified
```

gismo07, 2020-10-14:
Ok maybee we should check if your docker setup is working properly.
Can you 'see' your gpus when issuing:
```bash
sudo docker run --rm --gpus all nvidia/cuda:11.0-base nvidia-smi
```

now you should see something like this:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.51.06    Driver Version: 450.51.06    CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Tesla T4            On   | 00000000:00:1E.0 Off |                    0 |
| N/A   34C    P8     9W /  70W |      0MiB / 15109MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
```

zchrissirhcz, 2020-10-14:
@gismo07 Yes, I can correct run and get result for command `sudo docker run --rm --gpus all nvidia/cuda:11.0-base nvidia-smi`.  (Now I'm not with my computer but I remember I can run that)

zchrissirhcz, 2020-10-14:
On my ubuntu16.04, it is also nvidia 450 driver which support cuda 11.0

zchrissirhcz, 2020-10-15:
On my ubuntu16.04, first run this:
```bash
xhost local:root
```
then run this:
```bash
sudo docker run \
  -e SDL_VIDEODRIVER=x11 \
  -e DISPLAY=$DISPLAY\
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -p 2000-2002:2000-2002 \
  -it \
  --gpus all \
  carlasim/carla:0.9.10 ./CarlaUE4.sh -opengl
```
Then the CARLA window appears.

lz1159435992, 2020-10-15:
(base) [root@localhost ~]# sudo docker run --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=1 -e SDL_VIDEODRIVER=x11 -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -p 2000-2002:2000-2002 -it --gpus all carlasim/carla:0.9.10 ./CarlaUE4.sh -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found

Thank you, but it's not work for me. I got the same error. I guess that maybe path has some problems.

zchrissirhcz, 2020-10-15:
@lz1159435992 I am not using `root` user, but a sudo user.  From your pasted content, you are logged in as `root` user. Do you have a sudo user? If that case, may try it.

zchrissirhcz, 2020-10-15:
@lz1159435992 Btw, I'm using a ubuntu desktop, instead of a SSH connected terminal.

lz1159435992, 2020-10-15:
Maybe this is why I have so many problems. I have faced uncountable errors since I began to configure Carla on the server without display.

zchrissirhcz, 2020-10-15:
Well I think SSH to a remote linux server, which is with NVIDIA GPU, is a common required stuff.

I'm on windows10, and with [Xming](https://sourceforge.net/projects/xming/) and putty, I can get a DISPLAY. With this, commands like `xclock` can be displayed in my Windows.

However I still don't know how to let CARLA window display in Windows.

gismo07, 2020-10-15:
> Maybe this is why I have so many problems. I have faced uncountable errors since I began to configure Carla on the server without display.

Hmm I think, that running carla requires a display. I was also struggling at that point. 

Have you tried this: https://github.com/carla-simulator/carla/issues/225

lz1159435992, 2020-10-15:
> > Maybe this is why I have so many problems. I have faced uncountable errors since I began to configure Carla on the server without display.
> 
> Hmm I think, that running carla requires a display. I was also struggling at that point.
> 
> Have you tried this: #225

OK，I will try this, if you fix this please let me know how.

thytless, 2020-10-15:
`docker run -d -it --runtime=nvidia --name carla-server -e NVIDIA_VISIBLE_DEVICES=1 carlasim/carla:0.9.10.1 /bin/bash`
`docker exec -it carla-server bash`
`SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl `

Above works for me.

lz1159435992, 2020-10-15:
> Well I think SSH to a remote linux server, which is with NVIDIA GPU, is a common required stuff.
>
> I'm on windows10, and with [Xming](https://sourceforge.net/projects/xming/) and putty, I can get a [...]

There are little tutorials for doing this.  I have spend two weeks but still can‘t fix it.

lz1159435992, 2020-10-15:
> SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl

carla@bed76db766e3:~$ SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default

I got this error. It's new for me.

zchrissirhcz, 2020-10-15:
> There are little tutorials for doing this. I have spend two weeks but still can‘t fix it.

I think opengl's **indirect rendering** caused the failure of CARLA while `xclock` can start.

To be precise, I'm using VcXsrv software (Xlaunch) for X11 Service, 

[image]

[image]

And `glxinfo | grep ':'` gives:
[image]

which means I cannot start a program that using opengl4.6

Related links:
- https://community.khronos.org/t/opengl-version-is-forced-to-1-4-but-card-can-support-4-6-0/104868/3

I've tried adding my linux user to `video` group, as said in the pasted reference page, re-login shell, but `glxinfo`'s output remains.

thytless, 2020-10-15:
@lz1159435992 

> I got this error. It's new for me.

`docker exec -u root -it ${CARLA_CONTAINER_NAME} bash -c 'echo pcm.\!default { type plug slave.pcm "null" } > /etc/asound.conf'`
see #2820

lz1159435992, 2020-10-17:
I can't fix this. I have tried all the methods mentioned above. I try to install carla on windows10 with screen.

zchrissirhcz, 2020-10-17:
@lz1159435992  Yes, I think install and run CARLA on Windows 10 with screen is the best choice.

closed 2021-01-26 by germanros1987

alaamohamed2, 2021-04-17:
https://github.com/lz1159435992 Have you solved this issue?

dhiegomaga, 2021-05-10:
> > Maybe this is why I have so many problems. I have faced uncountable errors since I began to configure Carla on the server without display.
> 
> Hmm I think, that running carla requires a display. I was also struggling at that point.
> 
> Have you tried this: #225

I thought whole point of using Docker is not to need a display, if you are running on a remote server for instance:

> This tutorial is designed for:
> - **People who don't need to render the full simulation (the server is headless).**

https://carla.readthedocs.io/en/0.9.11/build_docker/

dhiegomaga, 2021-05-10:
Doing:

`sudo nvidia-docker run -it --rm  carlasim/carla:0.9.10 /bin/bash`

and then:

`SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl`

I get the same results as @lz1159435992: 

> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.
> sh: 1: xdg-user-dir: not found
> ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
> ...

But maybe this is normal? When checking `nvidia-smi` it does appear to be using the card. However, I am unable to connect to the carla-server using a local client... 

> Traceback (most recent call last):
>   File "set_environment.py", line 100, in main
>     world = client.load_world(args.town)
> RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000