carla-simulator/carla #3432: A question about openstreetmap

yangyueren, 2020-10-12:
Hi:
I'm new to use carla and UnrealEngine and I want to use carla 0.9.10 in the docker and import openstreetmap as my own map.
In sum, ubuntu headless server + docker(nvidia-docker2)+ openstreetmap.

I came across several questions.
1. In the tutorial https://carla.readthedocs.io/en/latest/build_linux/, I found that I should build UnrealEngine in my ubuntu server.
But if I download the below link and just start UE4Carla.sh,  I haven't used UnrealEngine. What's the difference?
https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.10.1.tar.gz

2. If I want to import the openstreetmap into carla, I should choose which way to start carla server?
https://carla.readthedocs.io/en/latest/build_linux/    or   just download the release version of carla https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.10.1.tar.gz

sagar-g-v, 2020-10-14:
Hi @yangyueren,

1. Build Unreal from source is not required unless you want modify and create your own package
    To use Docker image goto https://hub.docker.com/r/carlasim/carla.

    You can also use Carla Build Package  https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.10.1.tar.gz
     
2. For instaructions on importing `openstreetmap` in carla

    Please Refer Below link

    https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/

yangyueren, 2020-10-20:
> Hi @yangyueren,
> 
> 1. Building Unreal from source is not required unless you want modify and create your own package
>    To use Docker image goto https://hub.docker.com/r/carlasim/carla.
>    You can also use Carla Build Package  https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.10.1.tar.gz
> 2. For instructions on importing `openstreetmap` in carla
>    Please Refer Below link
>    https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/

Thanks for your reply. I use the Carla Build Package and everything goes well.

closed 2020-10-20 by yangyueren