carla-simulator/carla #3436: how to understand the data from the call world.get_level_bbs(carla.CityObjectLabel.Vehicles)

tinmodeHuang, 2020-10-13:
for example,`<[<carla.libcarla.BoundingBox object at 0x0000016E929F9AB0>,` how can I translate it into the content including position & rotation according to the description about `get_level_bbs` from official documentation? is there the easier way instead to take position & heading & velocity of surrounding vehicles?

germanros1987, 2021-01-26:
@doterop could you provide an answer for the posterity?

doterop, 2021-01-27:
Hi @tinmodeHuang 

The index of each Traffic Light returned by `world.get_level_bbs(carla.CityObjectLabel.TrafficLight)` will remain the same in the same simulation but this will only happen because new traffic lights will not be created.
The same logic applies to object types like `Buildings` or `Roads`, they will keep the same index.

A special case is `Vehicles` and `Pedestrians` object types. You can add or remove these objects at any point of the simulation altering this way the order and therefore the index will possibly change. But if you don't create or delete any objects of this type the index will keep the same.

Moreover, this index is only valid for the same simulation, it's not guaranteed to be the same between simulations. If you need to identify no spawnable objects like traffic lights you could use `world.get_environment_objects(carla.CityObjectLabel.TrafficLight)` which returns a list of `carla.EnvironmentObject`s. Each `EnvironmentObject` has a unique id that is valid between simulations and also had the bounding box of the object.

closed 2021-03-23 by doterop