carla-simulator/carla #3441: Supporting Ubuntu 20.04 on dist.carla.org

hofbi, 2020-10-13:
Are there any plans when the focal release will be added to dist.carla.org.

Today I tried a clean build of the Unreal Engine and Carla on 20.04 and it passed without any issues. I followed the steps in the [docs](https://carla.readthedocs.io/en/latest/build_linux/) for Ubuntu 18.04. Other than for 18.04. I did not install any python 2 dependencies and used g++9 instead of g++7. If there are more open question I am open to support here.

JWHennessey, 2020-10-16:
I agree supporting 20.04 would be very helpful. 

@hofbi - I haven't been as successful getting Carla 0.9.10 to install. Specifically, I can't install clang-8 as I run into a unmet dependency issue. 

`clang-8 : Depends: libllvm8 (>= 1:8~svn298832-1~) but it is not going to be installed
           Depends: libstdc++-5-dev but it is not installable
           Depends: libgcc-5-dev but it is not installable
           Depends: libobjc-5-dev but it is not installable
           Depends: libclang-common-8-dev (= 1:8.0.1+svn369350-1~exp1~20200112113617.82) but it is not going to be installed
           Depends: libclang1-8 (= 1:8.0.1+svn369350-1~exp1~20200112113617.82) but it is not going to be installed
           Recommends: llvm-8-dev but it is not going to be installed
           Recommends: libomp-8-dev but it is not going to be installed
 lld-8 : Depends: libllvm8 (= 1:8.0.1+svn369350-1~exp1~20200112113617.82) but it is not going to be installed
`

Did you have this issue and find a solution? It looks like the problem is with the package libffi6. 

I've tried installing with clang-10 but also run into problems. 

Thanks

hofbi, 2020-10-16:
@JWHennessey No I did not have this issue. But sounds like a circular dependency. Maybe try to remove everything related to clang and run an `apt autoremove`.

jhermen-List, 2020-10-21:
Having exactly the same issues on 20.04.
I installed g++9 and clang-9 as clang8 was uninstallable:

`The following packages have unmet dependencies:
 clang-8 : Depends: libllvm8 (>= 1:8~svn298832-1~) but it is not going to be installed
           Depends: libstdc++-5-dev but it is not installable
           Depends: libgcc-5-dev but it is not installable
           Depends: libobjc-5-dev but it is not installable
           Depends: libclang-common-8-dev (= 1:8.0.1+svn369350-1~exp1~20200112113617.82) but it is not going to be installed
           Depends: libclang1-8 (= 1:8.0.1+svn369350-1~exp1~20200112113617.82) but it is not going to be installed
           Recommends: llvm-8-dev but it is not going to be installed
           Recommends: libomp-8-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`

installing UnrealEngine and starting it worked fine, but when running the Carla make PythonAPI it complainsa bout the missing clang8

jhermen-List, 2020-10-21:
OK, got it working. 
The problem was caused by the Xenial repo that was added using the old build instructions. 

After removing "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" from /etc/apt/sources.list and running apt-get update i was able to install clang-8 

make PythonAPI and make launch

EsquilliDev, 2020-10-21:
> OK, got it working.
> The problem was caused by the Xenial repo that was added using the old build instructions.
>
> After removing "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" from [...]

I was having the same issue and this indeed fixed it. Thanks @jhermen-List

hofbi, 2020-12-25:
Are there any further plan with the new release?

CBalaji, 2021-01-03:
I am currently running ubuntu 20.10 and i am unable to get either carla(18.04) or ros(noetic 20.04) natively if i add the repo by faking  `$ lsb release -sc `  as bionic or focal respectively    i get unmet dependencies error , Ideally what should i do to get both running with the bridge ,
I have a new hardware and distros with kernel 5.8 only boot well

germanros1987, 2021-01-26:
@joel-mb could you comment on this?

joel-mb, 2021-02-24:
The latest CARLA release (0.9.11) for Ubuntu 20.04 has been added to the CARLA apt repository. Please see: https://github.com/carla-simulator/carla/issues/3374#issuecomment-785255690

closed 2021-02-24 by hofbi

fu1199, 2021-06-17:
I tried to build carla with Ubuntu 20.04 with following dependencies:

- sudo apt-get install build-essential clang-8 lld-8 g++-9 cmake ninja-build libvulkan1 python3-dev python3-pip libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev git
-pip3 install --user -Iv setuptools==47.3.1 && pip3 install --user distro
- sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 && sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
- pip3 install --user -Iv setuptools==47.3.1 &&
- pip3 install --user distro

however i run into the following problems:
.........
.........
/mnt/storage/carla/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:114:9: error: no member named 'strtold' in the global namespace
using ::strtold;
      ~~^
/mnt/storage/carla/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:115:9: error: no member named 'strtol' in the global namespace
using ::strtol;
      ~~^
/mnt/storage/carla/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:117:9: error: no member named 'strtoll' in the global namespace
using ::strtoll;
      ~~^
/mnt/storage/carla/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:119:9: error: no member named 'strtoul' in the global namespace
using ::strtoul;
      ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

has someone experienced the same problems? Can someone give advisory?

For me installing the carla package is not suitable because I want to adjust the Unreal Models which is then in my opinion not given.

Thanks a lot in advance.

sashajenner, 2021-09-01:
I get the same error

marcusvinicius178, 2022-01-13:
For me the the @jhermen-List  approach, commenting the line: "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" from /etc/apt/sources.list

Later running  these command lines below, worked:

`sudo  apt-get update`
`sudo apt-get install clang-8`

I have also added later this command line from CARLA tutorial, that this time runned without error:
`sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180`

I also needed to install this compiler
`sudo apt-get install ninja-build`

iscanGH, 2022-01-20:
> sudo apt-get install clang-8

@jhermen-List, could you elaborate more? My 'etc' folder is in anaconda3 and doesn't have any apt folder in it, so there is nothing to remove...

iscanGH, 2022-01-20:
I removed that line and only entered 
`sudo apt-get update &&
sudo apt-get install wget software-properties-common &&
sudo add-apt-repository ppa:ubuntu-toolchain-r/test &&
wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key|sudo apt-key add -`

then
`sudo apt-get update`

But when I try to install clang 8 still the same error appear
`The following packages have unmet dependencies:
 clang-8 : Depends: libllvm8 (>= 1:8~svn298832-1~) but it is not going to be installed
           Depends: libclang-common-8-dev (= 1:8.0.1+svn369350-1~exp1~20200114191400.80) but it is not going to be installed
           Depends: libclang1-8 (= 1:8.0.1+svn369350-1~exp1~20200114191400.80) but it is not going to be installed
           Recommends: llvm-8-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`

I run it on Ubuntu 20.4

marcusvinicius178, 2022-01-20:
Hi, @iscanGH did you try the command:

`sudo apt-get install aptitude`
`sudo apt aptitude install clang-8`

The aptitude is able to provide you with broken package solutions. Sometimes works sometimes not...it depends if your system is too bugged or not....libllvm8 is a dependency so you must install it to be able to install clang-8. It aptitude and the installation of the dependencies does not work, unfortunately, you need a fresh Linux system...

iscanGH, 2022-01-20:
@marcusvinicius178, I don't have any folder named 'etc'. Mt 'etc' folder is in anaconda3, but it doesn't contain 'apt' folder.

marcusvinicius178, 2022-01-21:
Hi @iscanGH every Linux has an etc folder....
Try deactivate conda. Maybe you are inside a virtual environment and cannot access this folder.

Just do:

`conda deactivate`

Then go to the root directory using:

`cd ..`

Then type:

`cd /etc/apt`

Then take a look in the files you have inside this folder:

`ls`

You will check that you have the sources_list file. Because the apt is from the command "sudo apt-get repo" this is the list containing the repositories which you can download packages from Ubuntu....if you didn't have this source_list file you wouldn't be able to download anything with sudo apt-get...therefore you have....just need to find where it is (should be under the /etc/apt/ folder) 

Then you finde the source_list:

`sudo gedit sources_list`

And then you need to locate and remove or comment the following line:

`deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main`

After that save this file:
Run:

`sudo apt-get update`

Open a new terminal and now try

`sudo apt-get install clang-8`

You will see that the broken packages issue does not happen anymore. Because those wrong repo (the line you must remove in sources_list) won't exist anymore and won't bring conflict issues.....

There is not other way to fix this issue...I remembered now that I used aptitude and also was not enough. Good luck.

chrninja, 2022-01-21:
@marcusvinicius178 I was able install clang-8 with regular `sudo apt-get install clang-8` and beforehand I removed the repository for the bionic clang deb with `sudo add-apt-repository -r "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main"`.

However when rebuilding UE4 I am noticing it is using clang-10 and not clang-8.

MrLerry, 2022-03-07:
Hi,
I've installed Carla on a Ubuntu 20.04 machine before, but I'm getting a new error when building from source with "make PythonAPI". Its on a new set up PC.
I was able to install clang8 with the solution @jhermen-List provided but im getting a CMake error now:

Resolving deltas: 100% (24/24), done.
Setup.sh: Compiling libc++.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- The ASM compiler identification is Clang
-- Found assembler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/clang++-8"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/XXX/CARLA/carla/Build/llvm-8.0-build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/ninja cmTC_e32c5 && [1/2] Building CXX object CMakeFiles/cmTC_e32c5.dir/testCXXCompiler.cxx.o
    [2/2] Linking CXX executable cmTC_e32c5
    FAILED: cmTC_e32c5 
    : && /usr/bin/clang++-8     CMakeFiles/cmTC_e32c5.dir/testCXXCompiler.cxx.o  -o cmTC_e32c5   && :
    /usr/bin/ld: cannot find -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:39 (project)

-- Configuring incomplete, errors occurred!

Did someone have a similar Issue before? Cant find anything for this Problem.
Thx in advance.

Alphabaijinde, 2022-03-09:
> OK, got it working. The problem was caused by the Xenial repo that was added using the old build instructions.
>
> After removing "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" from [...]

@[hofbi](https://github.com/hofbi)@[jhermen-List](https://github.com/jhermen-List) url has invalid what  "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-7 main" from "How to build CARLA on Linux"

shwetha-byte, 2022-06-27:
> Are there any plans when the focal release will be added to dist.carla.org.
>
> Today I tried a clean build of the Unreal Engine and Carla on 20.04 and it passed without any issues. I followed the [...]

Hi in my pc not installing clang -8

brilianputraa, 2022-10-19:
> Hi in my pc not installing clang -8

Hi, I also encountered the same issues when I tried to build Carla on Ubuntu 20.04. I fixed it by installing the clang-8 using this command:

`sudo apt-get install clang-8 lldb-8 lld-8`

You can refer to this site for looking more about the llvm package installation ([https://apt.llvm.org/](https://apt.llvm.org/))

roshambo919, 2023-10-05:
To solve the
```
  The C++ compiler

    "/usr/bin/clang++-8"

  is not able to compile a simple test program.
```
issue, I had to run `sudo apt install libstdc++-12-dev`