carla-simulator/carla #3442: Traffic Manager bottleneck for simulation speed updates

pandyarohan, 2020-10-14:
I have been experimenting with the synchrony and time-step settings to get faster than real-time simulation speeds to accelerate my experiments. The best way to do that that I have found is to vary the delta_time_step parameter to 0.05 seconds (highest recommended [here](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/)). 

However, weird things happen when I use traffic_manager with it. For one or two vehicles spawned within 20 m of each other, I still see 10x simulation speed. But as soon as I spawn a third, all the way to 20+ vehicles, the simulation speed becomes real time. This drop in speed is not affected by the number of vehicles spawned (3 vs 20). The problem also goes away once I disable the autopilot on those vehicles, leading me to believe that traffic manager is the bottleneck. 

I get about 23 frames per second, with 3 or more vehicles, regardless of the delta_time_step parameter in synchronous mode. I made a quick jupyter-notebook to test this that I have attached below. I read all the documentation on TrafficManager and Synchrony and time step, and utilitized all the advise here. 

The summarized question is: With all other things same, why does having more than two vehicles governed by traffic manager significantly slow down the simulation? Please let me know if I may be missing something, conceptually or if there is a bug? 
[lane_following_scenario.ipynb.zip](https://github.com/carla-simulator/carla/files/5375072/lane_following_scenario.ipynb.zip)

thytless, 2020-10-14:
According to my exp, traffic manager **before 0.9.10** DOES have a good impact on simulation speed when you have many vehicles in synchronous mode . It takes about 24ms for tm to update 75 vehicles on Town06 every sync tick.

What is your carla version? The new 0.9.10 edition claims that tm has a significant speedup, and what about having a test :)

tinmodeHuang, 2020-10-14:
hi @thytless. from my view of point, it looks like there's some toggle to break the limit for fps in the client. for example, it was point out that 'Use less CPU' being on lead to low performance in the item running Carla in F.A.Q in documentation.
would you have any idea what is to limit fps in the client that makes me feel it takes a bit longer to render of next frame?

thytless, 2020-10-14:
@tinmodeHuang 
1) Unfortunately I have not tried 0.9.10 since I failed to run released image on my machine, and so far I have no idea about the toggle to "break the limit for fps in the client"
2) Did you mean why it takes such long time to run a single tick? Too many sensors, too many vehicles controlled by traffic manager... there are a lot leading to your feeling about "lagging". This issue is highly recommended  https://github.com/carla-simulator/carla/issues/2617

jackbart94, 2020-10-14:
Hi,

> I have been experimenting with the synchrony and time-step settings to get faster than real-time simulation speeds to accelerate my experiments. The best way to do that that I have found is to vary [...]

You can now also put whatever delta_time_step you want, we need to update the documentation since this issue has been solved!

> However, weird things happen when I use traffic_manager with it. For one or two vehicles spawned within 20 m of each other, I still see 10x simulation speed. But as soon as I spawn a third, all the [...]

It makes sense that if you spawn more vehicles, the simulation becomes slower. More things to compute lead to decrease in performance. We are constantly trying to improve the efficiency of Traffic Manager, so this feedback is very important for us! By any chance, when you spawn the 3rd vehicle, is it still close to the other two?

> The summarized question is: With all other things same, why does having more than two vehicles governed by traffic manager significantly slow down the simulation? Please let me know if I may be missing something, conceptually or if there is a bug?

This thing about having more than two vehicles being a huge bottleneck is quite intriguing, we will run some tests to see if that's the case (and of course, you can do the same),

tinmodeHuang, 2020-10-14:
> @tinmodeHuang
>
> 1. Unfortunately I have not tried 0.9.10 since I failed to run released image on my machine, and so far I have no idea about the toggle to "break the limit for fps in the client"
> 2. [...]

1. thanks for letting me know about improving fps is so hard for the Carla team.
2.as for the toggle above I just discussed, more detail as follow:

> Running CARLA
 Low FPS rate when running the server in Unreal Editor.
UE4 Editor goes to a low performance mode when out of focus.
Go to Edit/Editor Preferences/Performance in the editor preferences, and disable the "Use Less CPU When in Background" option.

the above borrowed from the Documentation.
but I haven't found the exact place to enable the option

pandyarohan, 2020-10-14:
Hi @jackbart94, 

Some more information, I have CARLA 9.9, running on a  i7-9700K CPU @ 3.60GHz × 8 with a TITAN V. 

> It makes sense that if you spawn more vehicles, the simulation becomes slower. More things to compute lead to decrease in performance. We are constantly trying to improve the efficiency of Traffic [...]
I am trying to learn a lane following behaviour, so yes the third vehicle is also within 20m of the second one, and 40m from the first one all in the lane. 

> This thing about having more than two vehicles being a huge bottleneck is quite intriguing, we will run some tests to see if that's the case (and of course, you can do the same).
I ran many runs (not sure if 'tests' mean something more rigorous) with different vehicles 1-20+ being spawned on the same road and different delta_time_step (see notebook attached above). When the environment has 2 or fewer vehicles, the change in delta_time_step has a linear change in simulation speed. After 2 vehicles (could be 3 or 30) the simulation outputs 23 frames per second regardless of what I do. This is what points to it being a bug rather than slowdown due to additional computation. 

@thytless 
> According to my exp, traffic manager **before 0.9.10** DOES have a good impact on simulation speed when you have many vehicles in synchronous mode . It takes about 24ms for tm to update 75 vehicles on Town06 every sync tick.
I was on 9.9, I will give 9.10 a shot and update everyone on the findings.

closed 2021-04-27 by stale[bot]

qhaas, 2021-05-23:
CARLA 0.9.11's TM appears to be single-threaded (i.e. only using 'up to' 100% of a single CPU core) on my 4-core i7 system.  I haven't done profiling yet, but one can get CPU bottlenecked rather quickly.