carla-simulator/carla #3443: no light in camera sensor data

sagar-g-v, 2020-10-14:
Hi,
I am trying to read and save image from camera sensor

but i am getting images without light
- With Carla 0.9.10.1

[image]

Code Used for fetching the image from camera.
```python
@staticmethod
def _parse_image(weak_self, image):
    self = weak_self()
    if not self:
        return
    image.convert(ColorConverter.Raw)
    array = np.frombuffer(image.raw_data, dtype=np.dtype("uint8"))
    array = np.reshape(array, (image.height, image.width, 4))
    array = array[:, :, :3]
    array = array[:, :, ::-1]
    self._image_data = array
    if type(self._image_data) == np.ndarray:
        image = np.copy(self._image_data)
        if len(image.shape) < 3 or image.shape[2] == 1:
            image = cv2.cvtColor(image, cv2.COLOR_GRAY2RGB)
        else:
            image = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)

        cv2.imwrite('image.jpg', image)
```

But for 0.9.9.4 and for the same Code I am getting image with correct lightling

- With Carla 0.9.9.4
[image]

What could be the cause of this?
Is there any changes in way the image data is genrated in new version?
How is it working when rendered on pygame surface?

Please help me with this.

Madecu, 2020-10-14:
Hi @sagar-g-v,

In the new 0.9.10, we have changed the exposure mode of the camera to auto exposure and we have tweaked the camera configuration a little bit. 
We have exposed a lot of parameters so you can customize your RGB camera the same way you would in real life. You can set your camera (the RGB sensor of the car or the Spectator camera from unreal) with your customize ISO, aperture, shutter speed, and more advanced configurations. Check out the documentation, we listed all the parameters for the RGB sensors:
https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera

sagar-g-v, 2020-10-14:
@Madecu Thank you for the quick answer

I will try to adjust these new parameters and check.

Regards,
Sagar

sagar-g-v, 2020-10-14:
Hi @Madecu,

After adjusting exposure parameters, everything is working fine.

[image]

Thank you so much for your suggesion.

closed 2020-10-14 by sagar-g-v

charlie-wartnaby, 2020-11-26:
A big consequence of this change is that the provided example (carla/PythonAPI/examples/manual_control.py) appears to be broken. Currently for me it gives a very-nearly black pygame window running under Windows, and an overexposed image under Linux. Something seems to be wrong with setting the camera defaults. I spent ages trying to figure out why I had a 'blank' pygame window in Windows, before figuring out it had a few non-zero RGB values, and it was the raw data from CARLA where the problem started.

This gives me some clues to follow -- adding this comment so that others Googling the problem will be led here.

Charlie Wartnaby, Applus IDIADA