carla-simulator/carla #3444: carla0.9.10-xdg-user-dir: not found

lz1159435992, 2020-10-14:
When executing 0.9.10 docker with

$ docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=1 carlasim/carla:0.9.10
The following warning is reported:
sh: 1: xdg-user-dir: not found

and the container shut down at the same time.
what should I do to fix this ?

zchrissirhcz, 2020-10-15:
You may try my solution

https://github.com/carla-simulator/carla/issues/3431#issuecomment-708786234

badsmoke, 2020-10-19:
does not work for me

zchrissirhcz, 2020-10-19:
@lz1159435992 Please consider close this issue if you already found a solution.

lz1159435992, 2020-11-09:
I change computer and finally fix this, but still don't know why.

closed 2020-11-09 by lz1159435992

pavanteja295, 2021-01-12:
> @lz1159435992 Please consider close this issue if you already found a solution.

Hello @zchrissirhcz. The problem still persists when using docker on headless server with vulkan library. Do you know if this can be fixed?

zchrissirhcz, 2021-01-13:
Don't know vulkan stuff... For me, if using opengl is OK and vulkan easy to crash then pick opengl one.

qhaas, 2021-07-01:
It is unlikely that the cause of the crash is due to `xdg-user-dir: not found`, see Issue #3514 

Quite the red herring