carla-simulator/carla #3447: [Question] Detect if a car is driving on the wrong side of the road

FranNetti, 2020-10-14:
Hi, is there a way to detect when the car is driving on the wrong side of the road? I initially thought that the best way was to use a lane invasion detector but this works only on roads where there is a solid lane marking between the two sides of the road. 
Is there a way already built in Carla to detect it?

glopezdiest, 2020-10-16:
Hey @FranNetti, we have something like that implemented in SR. A simplified version is this code here ([WrongLaneTest.zip](https://github.com/carla-simulator/carla/files/5389921/WrongLaneTest.zip))

The idea behind it is to keep getting the **heading of the waypoint** that represents your vehicle's location. With normal driving, the angle between your current waypoint and the previous one will be very small. However, when you **change to an opposite lane**, the **heading difference will be much larger** (theoretically, of 180º). At that time, you know you moved to an opposite lane (just beware, you have to differentiate between going out of your lane and back into it).

This approach doesn't work for junctions though, as there are many different roads there and when you exit the intersections, you have to do some additional calculus.

Let me know if this helps

FranNetti, 2020-10-18:
Yes, thank you. It was exactly what I was looking for, it helped a lot!

closed 2020-10-18 by FranNetti