carla-simulator/carla #3452: [0.9.10.1][Synchronous mode] Failed to call world.tick() in world callback

thytless, 2020-10-15:
```python
import carla
import time
client = carla.Client('localhost', 2000)
print("Now load map")
client.load_world("Town01")
world = client.get_world()

print("Now switch to sync mode")
old_settings = world.get_settings()
world.apply_settings(carla.WorldSettings(True, False, 0.01))

def callback(snapshot):
    print(snapshot.frame)
    world.tick()

print("Now set callback")
callback_id = world.on_tick(callback)

print("Now first tick")
world.tick()
time.sleep(11)

print("Now remove callback")
world.remove_on_tick(callback_id)

print("Now reset settings")
world.apply_settings(old_settings)

```

The script above runs smoothly in 0.9.9.4 but got `RuntimeError: time-out of 10000ms while waiting for the simulator` in 0.9.10.1
Server run in docker, client ubuntu 18.04. No server-client version mismatch.

germanros1987, 2021-01-26:
@thytless do you still have this problem?

thytless, 2021-03-03:
Excuse me but I have been away from this issue for quite a long time, 0.9.9.4 is good and stable for me. Maybe I would like to test it with latest carla later.

qhaas, 2021-05-24:
Seeing this intermittently in CARLA 0.9.11 when I have several vehicles using traffic manager (and a CPU bottleneck develops, see Issue #3442).  Oddly enough, it always reports a `time-out of 10000ms` regardless of what `carla.Client::set_timeout` is set to.

closed 2023-06-02 by stale[bot]