carla-simulator/carla #3455: Carla Stops to load or fails while launching after build

akshatamk, 2020-10-15:
Followed all the steps at windows Build to setup carla on windows VM
1.I’m trying to build Carla 0.9.10
2.Have all the prerequisites installed.
3.“make PythonAPI” cmd was built properly.
4.After successful execution of “make launch” cmd…the CarlaUE4 is launching with this error.
[image]
Onclick of OK-option…the editor is opening but the screen is blank no map is visible.
Please help!

nagasanjay, 2020-10-15:
Did you download the assets and kept it in `content` folder ? or did you run `update.sh`/`update.bat` ?

akshatamk, 2020-10-16:
I downloaded the asset and kept it in content/carla folder

nagasanjay, 2020-10-18:
> I downloaded the asset and kept it in content/carla folder

Did you extract it ?

germanros1987, 2021-01-26:
@nagasanjay is this problem still bothering you?

closed 2021-07-21 by stale[bot]