carla-simulator/carla #3462: Enum comparison in behavior_agent fails even if True

t27, 2020-10-16:
This is regarding the enum comparisons in the behavior_agent here, https://github.com/carla-simulator/carla/blob/31fff28bc6d24ddae0b3977f8ac343ed16019202/PythonAPI/carla/agents/navigation/behavior_agent.py#L271

I am on Python 3.7 and Carla 0.9.9.4, 

The enum comparison doesnt work directly,

for example `self.direction == RoadOption.LANEFOLLOW` on line 286 is always `False`, even if the direction is indeed `LANEFOLLOW`

I have to use the `.value` attribute of the enum to ensure it works correctly,

Here's a screenshot from my debugging console,

[image]

glopezdiest, 2020-10-19:
Hey @t27. I'm using Python 3.6 and CARLA 0.9.10.1 and that check is indeed failing. However, CARLA also has that type of checks at the *global_route_planner* and the ScenarioRunner repository, and this isn't happening in any of those cases.

We'll take a look at it. Also, I'm tagging @jackbart94 .

raymanfx, 2020-11-30:
I noticed this issue as well. The behavior agent is completely broken in all recent CARLA releases due to this.
My workaround:

Remove the `RoadOption` class from `PythonAPI/carla/agents/navigation/local_planner_behavior` and import that enum from the `local_planner.py` class instead:

`from agents.navigation.local_planner import RoadOption`

germanros1987, 2021-01-26:
@glopezdiest could you please work on fixing this?

glopezdiest, 2021-01-27:
Yes, indeed

PR #3833 (2021-01-29): Fixed bug related to RoadOptions
### Description

Fixed a bug at the BehaviourAgent causing the RoadOptions to not work as intended.

This was caused by having two instances of the RoadOption enumerator, one at the *local_planner.py* and another one at the *local_planner_behavior.py*. Despite the fact that the latter file has one, this same file also uses the *global_route_planner.py* to calculate the route of the agent, which in turn uses the RoadOptions at the *local_planner.py*. This causes some unintended equality checks between these two different RoadOption enums, which will always be Fa [... cut]
closed 2021-02-15 by jackbart94