carla-simulator/carla #3464: Questions about infrustructure based simulation

zwbai, 2020-10-17:
Hi,

Thank you all for developing such a nice simulator for autonomous vehicle. Here I have a couple questions about the infrustructure-based driving simulation:

1) are the sensor modules (like camera, Lidar etc.) individual from host vehicle? For example, can I just put a camera and a Lidar sensor on the trffic light of a intersection and collect the sensing data directly? 
2) does CARLA proivde some communication modules? For example, can I use a road-side-unit (RSU) to send the collecting data to the target vehicles (vehicles with special flags).

Hope that someone can tell me the answer. I would really appreciate it!

Thanks,
Zhengwei

lynnpepin, 2020-10-23:
I can answer Question 1.

Carla sensors are created individually from vehicles, i.e. they are both separate actors.

A typical example of creating a camera might look like this:

```
camera_bp = blueprint_library.find('sensor.camera.rgb')
camera_transform = carla.Transform(carla.Location(x=1.5, z=3.5))
camera = world.spawn_actor(rgb_sensor_bp, camera_transform, attach_to=vehicle)
```

Here, `carla.Transform` is used to define a location and rotation for a camera. When we spawn the camera, we optionally attach it to an existing actor (in this case, a vehicle) so that it the camera will rotate and move with the attached actor.

If you want to anchor a camera to some arbitrary point in space, you will need to find the (x, y, z) coordinates of that point.

zwbai, 2020-10-25:
Hi @lynnpepin ,

Thank you very much for the answer and example.

So I am wandering that can I attach the camera to some infrustructures?, for instance,  traffic lights or some place near intersections.

germanros1987, 2021-01-26:
@zwbai in your use case you don't need to attach sensors to anything since traffic lights are static.

closed 2021-01-26 by germanros1987