carla-simulator/carla #3467: static link png, jpeg, and tiff libraries for client API users

qhaas, 2020-10-17:
A number of our researchers/developers that start using CARLA run into dependency issues because their environment lacks the appropriate png, jpeg, and tiff runtime libraries needed by the CARLA Client PythonAPI.  Not a show stopper, just takes an extra step to deploy them, unless they are running a non-supported distro in which we point them towards containers or the one we recompile in-house for RHEL7.

Assuming licensing permits (I'm no IP lawyer), it would speed things up if these libraries were statically linked into libcarla and thus have no runtime dependencies on them.  A side benefit is that it would work on more Linux distributions that lack the specific versions of these libraries linked when building on Ubuntu 18.04.

This would also reduce the number of issues posted here (e.g. Issue #3236  )

qhaas, 2020-12-23:
[Looks like libpng is being built from source now](https://github.com/carla-simulator/carla/blob/0.9.11/Util/BuildTools/Setup.sh#L388) (both static `.a` and dynamic `.so` libraries).  I'm no lawyer, but it looks like libpng is under a [permissive license](https://en.wikipedia.org/wiki/Zlib_License).  So, guess it is feasible to static link libpng instead of dynamically link it and thus avoid the additional dependency?

germanros1987, 2021-01-26:
That sounds about right, @qhaas

closed 2021-07-21 by stale[bot]