carla-simulator/carla #3472: I have logged the GNSS data into a file. How can I plot the same data on the Carla Town by reading logged file

manojp8, 2020-10-19:


manojp8, 2020-10-19:
[Vaidate_log.xlsx](https://github.com/carla-simulator/carla/files/5400130/Vaidate_log.xlsx)
Logged File

glopezdiest, 2020-10-20:
Hey @manojp8. in order to plot something you are gonna need CARLA's [debug tool](https://carla.readthedocs.io/en/latest/python_api/#carladebughelper). This tool however, uses the location to plot, not the geolocation, which is the one given by the GNSS. So before plotting, you need to change from one reference to another. Good news is that this is a recurring problem so there are several issues already created about how to do this.

PD: next time please, try to shorten the title to something like "How to plot GNSS data into CARLA" and explain more of the issue in the details. Also, it is always good to post your CARLA version to avoid possible confusions.

closed 2021-01-26 by germanros1987