carla-simulator/carla #3476: docker run headless nvidia not starting

badsmoke, 2020-10-19:
i just can't get carla to run on my server.
It is a headless server that can only be reached via ssh and that has a geforce 1660 ti installed.
all drivers are installed as you can see in the "nvidia-smi" output

What is the problem? docker was passed on the gpu, the rest should work out-of-the-box.

with carla version 0.9.5 you have even more debug output so i added this one.

The other Issues with similar errors I already searched through but none of the solutions helped.

my system
ubuntu 20.04
amd ryzen 3950X 
128 GB RAM
Geforce gtx 1660 Ti 6GB

```
sudo docker run --rm --gpus all nvidia/cuda:11.0-base nvidia-smi
Mon Oct 19 12:43:40 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.80.02    Driver Version: 450.80.02    CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 166...  Off  | 00000000:0D:00.0 Off |                  N/A |
| 35%   46C    P0    12W / 120W |      0MiB /  5936MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

```

docker run witht carla 0.9.10.1
```sudo docker run --rm -it --runtime=nvidia --name carla-server -p 2000-2002:2000-2002  -e NVIDIA_VISIBLE_DEVICES=0 -e SDL_VIDEODRIVER=offscreen carlasim/carla:0.9.10.1 /bin/bash ./CarlaUE4.sh -quality-level=Low -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
```

docker run with carla 0.9.5
```
sudo docker run --rm -it --runtime=nvidia --name carla-server -p 2000-2002:2000-2002 -e NVIDIA_VISIBLE_DEVICES=0 -e SDL_VIDEODRIVER=offscreen carlasim/carla:0.9.5 /bin/bash ./CarlaUE4.sh -quality-level=Low -opengl
Increasing per-process limit of core file size to infinity.
- Existing per-process limit (soft=18446744073709551615, hard=18446744073709551615) is enough for us (need only 18446744073709551615)
sh: 1: xdg-user-dir: not found
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogPakFile: Registered encryption key '00000000000000000000000000000000': 0 pak files mounted, 0 remain pending
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.058167
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +0:00, Platform Override: ''
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin CryptoKeys
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin MeshEditor
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin AlembicImporter
[... 374 lines cut ...]
Malloc Size=68112 LargeMemoryPoolOffset=199248 
Malloc Size=48848 LargeMemoryPoolOffset=248112 
[2020.10.19-12.47.29:313][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000010

[2020.10.19-12.47.29:313][  0]LogCore: Fatal error!

0x0000000007ad6308 CarlaUE4!UnknownFunction(0x78d6308)
0x0000000007aaed7e CarlaUE4!UnknownFunction(0x78aed7d)
0x0000000005c3744c CarlaUE4!_Z33Linux_PlatformCreateDummyGLWindowP22FPlatformOpenGLContext(+0x2b)
0x0000000005c3970d CarlaUE4!_Z18PlatformInitOpenGLv(+0x93c)
0x0000000005c581ca CarlaUE4!_ZN17FOpenGLDynamicRHIC1Ev(+0x2c9)
0x0000000005c57eec CarlaUE4!_ZN23FOpenGLDynamicRHIModule9CreateRHIEN16ERHIFeatureLevel4TypeE(+0x1b)
0x0000000005cee529 CarlaUE4!_Z24PlatformCreateDynamicRHIv(+0x418)
0x0000000005ced8c6 CarlaUE4!_Z7RHIInitb(+0x75)
0x0000000003ebfdca CarlaUE4!_ZN11FEngineLoop7PreInitEPKDs(+0x2c89)
0x0000000003ecb5f8 CarlaUE4!_Z11GuardedMainPKDs(+0x47)
0x000000000789acdb CarlaUE4!_Z14CommonUnixMainiPPcPFiPKDsE(+0x95a)
0x00007fadec63f830 libc.so.6!__libc_start_main(+0xef)
0x0000000003ebd029 CarlaUE4!UnknownFunction(0x3cbd028)

[2020.10.19-12.47.29:323][  0]LogExit: Executing StaticShutdownAfterError
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
./CarlaUE4.sh: line 5:    12 Segmentation fault      (core dumped) "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4" CarlaUE4 $@
```

qhaas, 2020-10-19:
`nvidia-smi` working in docker is a good sign.

I have had [mixed results](https://github.com/carla-simulator/carla/issues/225#issuecomment-533795704) with CARLA 0.9.5 and headless docker.  Try the following:
```
$ docker run -p 2000-2002:2000-2002 --gpus=all --name carla-server -d carlasim/carla:0.9.6
...
$ nvidia-smi pmon | grep CarlaUE4
    0      23533   C+G    35    14     -     -   CarlaUE4-Linux-
    1      23533   C+G     -     -     -     -   CarlaUE4-Linux-
    2      23533   C+G     -     -     -     -   CarlaUE4-Linux-
    3      23533   C+G     -     -     -     -   CarlaUE4-Linux-
```

badsmoke, 2020-10-19:
nvdia-smi time no reaction at all, because carla can't even continue to start

hellojql, 2020-10-23:
> found

I met the same problem with 0.9.10

hellojql, 2020-10-23:
> i just can't get carla to run on my server.
> It is a headless server that can only be reached via ssh and that has a geforce 1660 ti installed.
> all drivers are installed as you can see in the "nvidia-smi" output
> 
> What is the problem? docker was passed on the gpu, the rest should work out-of-the-box.
> 
> with carla version 0.9.5 you have even more debug output so i added this one.
> 
> The other Issues with similar errors I already searched through but none of the solutions helped.
> 
> my system
> ubuntu 20.04
> amd ryzen 3950X
> 128 GB RAM
> Geforce gtx 1660 Ti 6GB
> 
> ```
> sudo docker run --rm --gpus all nvidia/cuda:11.0-base nvidia-smi
> Mon Oct 19 12:43:40 2020       
> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 450.80.02    Driver Version: 450.80.02    CUDA Version: 11.0     |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |                               |                      |               MIG M. |
> |===============================+======================+======================|
> |   0  GeForce GTX 166...  Off  | 00000000:0D:00.0 Off |                  N/A |
> | 35%   46C    P0    12W / 120W |      0MiB /  5936MiB |      0%      Default |
> |                               |                      |                  N/A |
> +-------------------------------+----------------------+----------------------+
>                                                                                
> +-----------------------------------------------------------------------------+
> | Processes:                                                                  |
> |  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
> |        ID   ID                                                   Usage      |
> |=============================================================================|
> |  No running processes found                                                 |
> +-----------------------------------------------------------------------------+
> ```
> 
> docker run witht carla 0.9.10.1
> 
> ```
> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.
> sh: 1: xdg-user-dir: not found
> ```
> 
> and exit ....
> 
> docker run with carla 0.9.5
> 
> ```
> sudo docker run --rm -it --runtime=nvidia --name carla-server -p 2000-2002:2000-2002 -e NVIDIA_VISIBLE_DEVICES=0 -e SDL_VIDEODRIVER=offscreen carlasim/carla:0.9.5 /bin/bash ./CarlaUE4.sh -quality-level=Low -opengl
> Increasing per-process limit of core file size to infinity.
> - Existing per-process limit (soft=18446744073709551615, hard=18446744073709551615) is enough for us (need only 18446744073709551615)
> sh: 1: xdg-user-dir: not found
> LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
> LogPakFile: Registered encryption key '00000000000000000000000000000000': 0 pak files mounted, 0 remain pending
> LogPlatformFile: Using cached read wrapper
> LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
> LogStats: Stats thread started at 0.058167
> LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +0:00, Platform Override: ''
> LogPluginManager: Mounting plugin Paper2D
> LogPluginManager: Mounting plugin AISupport
> LogPluginManager: Mounting plugin LightPropagationVolume
> LogPluginManager: Mounting plugin AssetManagerEditor
> LogPluginManager: Mounting plugin CryptoKeys
> LogPluginManager: Mounting plugin FacialAnimation
> LogPluginManager: Mounting plugin MeshEditor
[... 382 lines cut ...]
> [2020.10.19-12.47.29:313][  0]LogCore: Fatal error!
> 
> 0x0000000007ad6308 CarlaUE4!UnknownFunction(0x78d6308)
> 0x0000000007aaed7e CarlaUE4!UnknownFunction(0x78aed7d)
> 0x0000000005c3744c CarlaUE4!_Z33Linux_PlatformCreateDummyGLWindowP22FPlatformOpenGLContext(+0x2b)
> 0x0000000005c3970d CarlaUE4!_Z18PlatformInitOpenGLv(+0x93c)
> 0x0000000005c581ca CarlaUE4!_ZN17FOpenGLDynamicRHIC1Ev(+0x2c9)
> 0x0000000005c57eec CarlaUE4!_ZN23FOpenGLDynamicRHIModule9CreateRHIEN16ERHIFeatureLevel4TypeE(+0x1b)
> 0x0000000005cee529 CarlaUE4!_Z24PlatformCreateDynamicRHIv(+0x418)
> 0x0000000005ced8c6 CarlaUE4!_Z7RHIInitb(+0x75)
> 0x0000000003ebfdca CarlaUE4!_ZN11FEngineLoop7PreInitEPKDs(+0x2c89)
> 0x0000000003ecb5f8 CarlaUE4!_Z11GuardedMainPKDs(+0x47)
> 0x000000000789acdb CarlaUE4!_Z14CommonUnixMainiPPcPFiPKDsE(+0x95a)
> 0x00007fadec63f830 libc.so.6!__libc_start_main(+0xef)
> 0x0000000003ebd029 CarlaUE4!UnknownFunction(0x3cbd028)
> 
> [2020.10.19-12.47.29:323][  0]LogExit: Executing StaticShutdownAfterError
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> ./CarlaUE4.sh: line 5:    12 Segmentation fault      (core dumped) "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4" CarlaUE4 $@
> ```
> 
> and exit ....

you can try the follows, and it works for me.
docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.10 bash -c "SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl"

qhaas, 2020-12-02:
Would be quite handy if we could figure out a way to get vulkan working with newer versions of CARLA in a headless environment, seems all the examples of running it without X involve using OpenGL.

pavanteja295, 2021-01-12:
hello @qhaas.  Were you able to figure out running docker with vulkan support on a headless server for carla0.9.11?

qhaas, 2021-01-26:
No, but I haven't spent allot of time chasing it down because OpenGL was working fine... until I started running into Issue #3377.  I fear vulkan might require an x-server...

germanros1987, 2021-01-26:
Let me provide a vulkan recipe:

* Make sure you have a Ubuntu 18.04 system
*Fetch the latest NVIDIA driver: ```wget http://download.nvidia.com/XFree86/Linux-x86_64/450.57/NVIDIA-Linux-x86_64-450.57.run```
* Install the driver: ```sudo /bin/bash NVIDIA-Linux-x86_64-450.57.run --accept-license --no-questions --ui=none```
* Install xserver related dependencies: ```sudo apt-get install -y xserver-xorg mesa-utils libvulkan1```
* Configure xserver
* ```sudo nvidia-xconfig --preserve-busid -a --virtual=1280x1024```
* ```sudo nvidia-xconfig --preserve-busid  --enable-all-gpus --use-display-device=None --virtual=1280x1024```
* Run xserver
```sudo X :0 &```
* Run CARLA: ```DISPLAY=:0.GPU ./CarlaUE4.sh -vulkan```
* alternative: Run CARLA in a docker:  ```sudo -E docker run --privileged --rm --gpus all -it --net=host -v /tmp/.X11-unix:/tmp/.X11-unix:rw docker-image:latest /bin/bash ```
* ```DISPLAY=:0.GPU ./CarlaUE4.sh -vulkan```

Notes:
* Make sure that within the container the env variable SDL_VIDEODRIVER is set to x11, i.e., SDL_VIDEODRIVER=x11
* Make sure that within the container libvulkan1 is installed

qhaas, 2021-01-26:
Thanks;  For our use case, we are hoping to avoid running an x-server, for we use non-interactive HPC systems and servers for long-term, batch carla runs.  I'm planning to build CARLA 0.9.11 against UE 4.25.4 to see if we can get the desired behavior.

I was unable to locate a specific commit that 'fixes' headless rendering with vulkan/linux/ue in Epic's logs, had hoped to backport only that fix to UE 4.23.1 for ppc64le and UE 4.24.x for x86-64.

closed 2021-07-21 by stale[bot]