carla-simulator/carla #3478: OpenDRIVE standalone mode: export of generated 3D mesh possible?

phzeller, 2020-10-19:
Hello together,

I was able to successfully open an OpenDRIVE file (.xodr) in OpenDRIVE standalone mode.
However, I was wondering if it is possible to export the generated 3D mesh (i.e. as .obj)?

Thank you so much and best regards!

XGodina, 2020-10-26:
You mean after importing the OpenDrive into CARLA and generating the mesh, then exporting that mesh ?

phzeller, 2020-10-27:
Exactly @XGodina

marcgpuig, 2020-10-27:
Hi @phzeller,
We indeed have a way to export this mesh as a [.obj](https://en.wikipedia.org/wiki/Wavefront_.obj_file) file however it's not exposed to the PythonAPI. It's in `carla/LibCarla/source/carla/geom/Mesh.cpp` and it's called `Mesh::GenerateOBJ()`. I have plans to expose the whole mesh system to Python but we have other priorities right now, so sadly this is not in our current scope.
As a workaround, you can just call this function from another part of the C++ code and save it to a file. When I did it, Blender 2.82 worked fine for me opening the file.

closed 2021-01-26 by germanros1987

Sephiroth1118, 2024-09-05:
> Hi @phzeller, We indeed have a way to export this mesh as a [.obj](https://en.wikipedia.org/wiki/Wavefront_.obj_file) file however it's not exposed to the PythonAPI. It's in [...]

Hi @marcgpuig , 
Sorry to bother you! 
I'm new to Carla and recently I'm working on a project that needs to add some building to map which we only have opendrive files loaded in standalone mode. So I want to save the 3D mesh generated by Carla then load it in UE editor and add some buildings.
I found this comment, but I haven't used the C++ API and am confused about how to use this function. Could you give me some advice, like how to get the mesh information of the server? 
Thank you very much!