carla-simulator/carla #3479: A question on fixed_delta_seconds of synchronous_mode

yangyueren, 2020-10-19:
Hi, I want to produce a video at fps 30 captured by the camera RGB sensor, but I am confused by the synchronous_mode of carla.

I started the carla server with the following command:
```
CARLA_EXEC=./CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping
$CARLA_EXEC -benchmark -fps=30 -quality-level=Epic -fixedseed=10 -opengl
```
Although it works, I don't know what is -fps used for?

In my client:
```
settings = world.get_settings()
settings.synchronous_mode = True
settings.fixed_delta_seconds = 0.04
settings.no_rendering_mode = True
world.apply_settings(settings)
```
If I want a video at fps 30, what should the fixed_delta_seconds set?

Last question, what's the difference between world.wait_for_tick() and world.tick()
```
   while True:
        try:
            world.tick()
        except:
            pass
```
I haven't used world.wait_for_tick() and is there anything wrong in my code?

Thanks!

glopezdiest, 2020-10-20:
Hey @yangyueren.

When a simulation is run in **asynchronous mode**, the server is the one making the tick, and all that ***world.wait_for_tick()*** does is block until the server has ticked.
On the other hand, in **synchronous mode**, you as the client are responsible of ticking the server, so the simulation won't advance until you do a ***world.tick()***.

What the *fixed_delta_seconds* does is set the time of the simulation that passes between ticks (either in async or sync mode). So, if you want **30 FPS**, you need a ***fixed_delta_seconds* of 1/30**. It is important to note that the sync mode doesn't have to run at real time and that that will depend on your calculus between ticks. That means that if you want to record a video, you won't see 30 FPS during simulation, but the video will be recorder at that FPS.

In not actually sure what the *-fps* argument does, but I'd assume that it sets the *fixed_delta_seconds* to the inverse of that value.

Let me know if that helps :slightly_smiling_face:

yangyueren, 2020-10-20:
> Hey @yangyueren.
>
> When a simulation is run in **asynchronous mode**, the server is the one making the tick, and all that _**world.wait_for_tick()**_ does is block until the server has ticked.
> On the [...]

Thank you for your prompt reply. With the explanation above, I have understood how the synchronous mode works. Thanks again.

closed 2020-10-20 by yangyueren