carla-simulator/carla #3482: `make import` throws ERROR: make: *** No rule to make target 'import'. Stop.

philippwulff, 2020-10-20:
Hello, I want to import my own Opendrive map into carla but the terminal fails for me.

I followed the instructions from the [doc](https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source) and placed my map.xodr and my map.fbx in the carla-simulator/Import/ folder.
According to the doc, I should next call `make import ARGS="--package package_name"`

When I run the command from the doc in the terminal in the folder /opt/carla-simulator/, I receive this error:

`make: *** No rule to make target 'import'. Stop.`

[This](https://github.com/carla-simulator/carla/issues/2022) issue seems to be the same, but is was not resolved. One answer said to execute the command in the root folder of the source code from carla, but that is what I already did before.

I am running Linux Mint 18.3 and installed CARLA from source.

philippwulff, 2020-10-21:
Also, I followed the installation process from the Quick Installation in den Doc.

closed 2020-10-22 by philippwulff

Mattiapzz, 2022-01-27:
@philippwulff Sorry to bother, did you solve that problem? If so, how? I am finding myself in the same position.

philippwulff, 2022-01-27:
@Mattiapzz Hi, I do not remember whether I solved this. But I believe the importing process was very buggy back when I was trying to use it. I am also not using Carla anymore...