carla-simulator/carla #3487: [TrafficManager] How to change autopilot speed?

sagar-g-v, 2020-10-20:
when `actor.set_autopilot(True)` is used the default speed limit is at 17kmph(~20kmh)

Is there any way to set/change autopilot speed?

zchrissirhcz, 2020-10-21:
```python
                        world.player.enable_constant_velocity(carla.Vector3D(7, 0, 0))
                        world.constant_velocity_enabled = True
```

See `example/manual_control.py` around  line 350 for reference.

jackbart94, 2020-10-21:
Hi,

> ```python
>                         world.player.enable_constant_velocity(carla.Vector3D(7, 0, 0))
>                         world.constant_velocity_enabled = True
> ```
> 
> See `example/manual_control.py` around line 350 for reference.

They are using Traffic Manager, so this is not the case.

> when `actor.set_autopilot(True)` is used the default speed limit is at 17kmph(~20kmh)
> 
> Is there any way to set/change autopilot speed?

Everything you need to know about setting or changing stuff in Traffic Manager is at [this link](https://carla.readthedocs.io/en/latest/python_api/#carlatrafficmanager). In particular, you might be interested in ``global_percentage_speed_difference`` and ``vehicle_percentage_speed_difference``

closed 2020-11-03 by sagar-g-v

AIasd, 2021-07-11:
Hi @jackbart94 , is there a way to set current speed limit globally? It is a bit inconvenient to use `global_percentage_speed_difference` and `vehicle_percentage_speed_difference` when the speed limit for roads change.