carla-simulator/carla #3488: Several buildings have multiple bounding boxes

ARSekkat, 2020-10-20:
I projected the nearby buildings bounding boxes of the vehicle, and as shown in the image attached the same building have multiple bounding boxes, it's the case in Town10HD too. Is this on purpose?
Ubuntu 18.04.5 LTS
CARLA 0.9.10.1 

[image]

Madecu, 2020-10-21:
Hi @ARSekkat, this happens because we built those buildings as procedurals, where we have different "pieces" of the building so we can create different shapes with the same meshes. So every bounding box you see, it is the bounding box of that particular "piece".
We use this method almost entirely to build Town10HD.

[image]

ARSekkat, 2020-10-21:
Hello @Madecu, Thank you for your answer.
Is there a way to have one bounding box for each building?

Madecu, 2020-10-21:
Well, it is more complex than it seems. If we have the bounding box from the 'group of meshes', depending on the shape of the building, it will have a lot of empty spaces. There are also other assets inside those 'groups' that will cause an inaccurate bounding box.
We are currently working on the information sent back from the bounding boxes. We will study the possibility of implementing it in these cases.

ARSekkat, 2020-10-21:
Ok, thank you.

closed 2020-10-21 by ARSekkat