carla-simulator/carla #3489: terminate called after throwing an instance of 'std::out_of_range' when calling blueprint_library

will-sloan, 2020-10-21:
I am running `tutorial.py` and the for loop at the end, which spawns 10 cars, causes the blueprint_library to sometimes throw the error: 
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
```
it doesn't happen every time which is strange. 
There are some warnings at the start which may be relevant: 
```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.10.1-30-g31fff28b 
WARNING: Simulator API version  = 0.9.10.1-30-g31fff28b-dirty
```

I saw another issue with a similar problem but it was with the traffic manager. 

Operating System: Ubuntu 18.04 LTS  
I followed the guide last week to build carla, so it should be the latest version. 

Also, is there a way to catch this error in python so that the `finally` clause runs to clean everything up?

jackbart94, 2020-10-23:
Hi,

Thanks for pointing this out, we will be looking into this issue and solve it!

thillRobot, 2020-10-30:
I am also having this issue with CARLA 0.9.10.1 in Ubuntu 18.04. It is strange because it seems to happen slightly differently each time. Every once and while it seems to work.

This time it did not work.

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 784d9b9f 
WARNING: Simulator API version  = 0.9.10.1 
created vehicle.chevrolet.impala
created sensor.camera.depth
moved vehicle to Location(x=-2.350990, y=-2.835118, z=-0.007039)
created vehicle.bmw.isetta
created vehicle.audi.etron
created vehicle.mustang.mustang
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
```

This time it did.

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 784d9b9f 
WARNING: Simulator API version  = 0.9.10.1 
created vehicle.harley-davidson.low_rider
created sensor.camera.depth
moved vehicle to Location(x=285.194214, y=17.596277, z=-0.028990)
created vehicle.bh.crossbike
created vehicle.audi.tt
created vehicle.bh.crossbike
created vehicle.seat.leon
created vehicle.dodge_charger.police
created vehicle.dodge_charger.police
created vehicle.harley-davidson.low_rider
created vehicle.tesla.model3
created vehicle.toyota.prius
created vehicle.citroen.c3
destroying actors
done.
```

Most of the time it does not.

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 784d9b9f 
WARNING: Simulator API version  = 0.9.10.1 
created vehicle.bmw.grandtourer
created sensor.camera.depth
moved vehicle to Location(x=-27.323112, y=0.536298, z=0.071206)
created vehicle.nissan.patrol
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted (core dumped)
```

The script `manual_drive.py` works for me, so that brings me to another question. What is `tutorial.py` supposed to do?

closed 2020-11-07 by jackbart94

qhaas, 2021-01-04:
Seems to be fixed in CARLA 0.9.11, thanks guys!

PR #3542 (2020-11-07): Registration lock while frame re-allocation
#### Description

Fixes a race condition that was happening when using the tutorial.py script.

Fixes #3492
Fixes #3489

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

None

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3542)