carla-simulator/carla #3490: CARLA 0.9.9 cannot launch

adityapatruni, 2020-10-21:
I have CARLA 0.9.9 installed on Ubuntu 20.04. I use NVIDIA GPU 1650 with 450.51 drivers. I have CUDA 10.1 installed.

I get the following error when I run ./CarlaUE4.sh in the /opt/carla-simulator/bin directory. How can I fix this? Thank you.

 4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x2200041
  Serial number of failed request:  386
  Current serial number in output stream:  397
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)

Unlingius, 2020-12-10:
@adityapatruni   if still matter, `./CarlaUE4.sh -opengl` worked for me

closed 2021-01-26 by germanros1987

lucasmazz, 2022-07-31:
I'm having the same error. But, when I try to execute with the -opengl flag it shows a warning message that says "OpenGL is no longer supported for desktop platforms, Vulkan will be used instead" and the application crashes the same way

ngonza27, 2022-10-10:
I'm having the same error as @lucasmazz

celik02, 2023-10-26:
I had the same problem, -opengl flag did not solve the issue. Hopefully -prefernvidia flag helped. 

`./CarlaUE4.sh -prefernvidia`