carla-simulator/carla #3493: Non determinism of Walker AI controllers

pavanteja295, 2020-10-22:
Hello everyone,
Thanks for making the Traffic-Manager deterministic in CARLA 9.10.1(#3333). However, the pedestrians with walker AI controllers still seem to behave in non-deterministic ways. This would prohibit users from repeating the simulation over multiple runs. 

It would be great if you can let me know if there is already a workaround for making walkers with AI controller **deterministic*. Can you let me know if there is a seed within the AI controller that can help me replicate the same simulation over multiple runs?

Thanks in advance.

pavanteja295, 2021-01-12:
Has this been fixed in carla 0.9.11?

germanros1987, 2021-02-23:
Hi @pavanteja295, sorry for the delay. @bernatx will come back to you with an answer soon.

Paniac, 2021-05-31:
Hi @bernatx, do you have any updates on this?

bernatx, 2021-06-01:
Hi, I will check if adding an specific seed for the Recast & Detour library is enough to get deterministic pedestrians, and I will get back to you. Regards

mzheng27, 2021-06-04:
Hi @bernatx, I also interested in the pedestrian reproducibility. I wonder where I should add the seed for the detour and recast git repo? Do you think it is necessary to modify other walker navigation files? Thank you!

qhaas, 2021-08-31:
We would find this feature useful as well.  Was it added in 0.9.12?

bernatx, 2021-09-14:
This feature is not included in 0.9.12 version but it will be in next release in a few weeks. 
You can use the **dev** branch version in the next days where it will be included.

Pyrestone, 2022-10-07:
@bernatx  I am still encountering this source of non-determinism in 0.9.13 Any updates on this?

closed 2023-06-02 by stale[bot]