carla-simulator/carla #3494: Fog doesn't work - black sky

PTorr, 2020-10-22:
I'm trying to create fog using environment.py --fog (also tried weather.py).
However all I get is black sky with no fog.
I've tried multiple configurations and towns but had no luck.
What can be the problem?

Madecu, 2020-10-22:
Hi @PTorr, do you have the latest version of CARLA? You might have the same issue that they have here https://github.com/carla-simulator/carla/issues/3426

PTorr, 2020-10-25:
Yes I do have the latest version.

PTorr, 2020-10-25:
Thanks for the referral.

Going back to version 9.9.4 fixed the problem for now.

I hope that in the next release things will go back to normal.

Keep making CARLA great 😊.

Torr.

corkyw10, 2021-02-24:
@Madecu could you follow up on this please?

Madecu, 2021-03-02:
Hi @PTorr,

Have you tried the new version of CARLA? 
The problem you had is it still happening?

We appreciate your feedback and see if we can help you

PTorr, 2021-03-02:
Still the same.

I do work in Ubuntu.

Madecu, 2021-03-02:
Can you send an image of the problem? If you don't change the fog values, the sky is still black or it becomes black when changing fog density?

PTorr, 2021-03-03:
[image]

When I'm not using fog parameter everything works fine.

Madecu, 2021-03-03:
I can't reproduce your error. I do see the fog in Linux. Can you show me how do you change the fog value?
I used environment.py --fog 100 in this example

[image]

PTorr, 2021-03-03:
exactly the same as you.
maybe because I'm using:`./CarlaUE4.sh -opengl`

Madecu, 2021-03-11:
Yes, I could reproduce the error using the -opengl tag. I don't know exactly why openGL doesn't render the fog.
The sky turns black because we have to simulate that very dense fog would hide the sunlight. It is the same with clouds, they are not volumetric, so they don't project shadows. 
In the next release we have updated unreal engine to 4.26 and it won't support opengl anymore. I recommend you to use vulkan instead.

PTorr, 2021-03-11:
Thanks for the detailed explanation.

closed 2023-06-02 by stale[bot]