carla-simulator/carla #3495: Can't compile RSS

Hwansoo-Choi, 2020-10-22:
Hello.

I'm trying to use RSS sensor but I have a problem with compiling it.

This is my environment info
-UBUNTU 16.04
-Carla 0.9.10.1 (I reinstalled it today)
-Python 3.8

I've tried what the page below says.

https://carla.readthedocs.io/en/latest/adv_rss/

But when I run
`make PythonAPI.rss`

next error message occurs

> gcc: error: /home/hwansoo/carla/PythonAPI/carla/dependencies/lib/libad_rss_map_integration_python38.a: No such file or directory

> gcc: error: /home/hwansoo/carla/PythonAPI/carla/dependencies/lib/libad_map_access_python38.a: No such file or directory

> gcc: error: /home/hwansoo/carla/PythonAPI/carla/dependencies/lib/libad_rss_python38.a: No such file or directory

> gcc: error: /home/hwansoo/carla/PythonAPI/carla/dependencies/lib/libad_physics_python38.a: No such file or directory

I've tried this with several different python versions, but I failed.

Can you help me to solve this problem please?

Always thank you for all your effort!

direcf, 2020-10-23:
Same here + 1

berndgassmann, 2020-10-29:
Hi,
did you try the current dev branch, too?
The PR #3388 should have solved python3 built issues.

If you want to make use of the CARLA 0.9.10.1 then you have to use the pre-built version, which is compiled only with python2.7 support (https://github.com/carla-simulator/carla/releases/tag/0.9.10.1). But maybe that one doesn't work for general library reasons under Ubunut16.04... at least that's what I fear.

Regards,
Bernd.

Hwansoo-Choi, 2020-10-29:
Yes I'm trying with the latest versions of dev branch.
I followed next steps.

`  git clone -b dev --single-branch https://github.com/carla-simulator/carla.git `
` cd carla `
` ./Update.sh `
` make PythonAPI `
` make launch `
` make LibCarla `
` make LibCarla.rss `
` make PythonAPI.rss `

But the last line makes errors I wrote above.

Is there any problem with my install procedure?

Thank you for reply!
Best regards, Hwansoo.

berndgassmann, 2020-10-29:
Instead of all the make commands a single
make package.rss 
for the initial build should be enough. I don't know if the previous build (make PythonAPI; make launch; and make LibCarla) which build without RSS lead to the problems you see.

Then you said you are using python3.8, is this the default python installation on your system (most probably the call of python --version will show the python 3.8 right?) Otherwise maybe the explicit selection of the python version could make sense:
make package.rss ARGS="--python-version=3.8"

Somehow it looks as if the install step of LibCarla is not installing the RSS python libs. As these are collected using a glob expression within CMake I have an idea what might happen:
The first run of make LibCarla is not finding the RSS libraries (as they are not yet built).
The second run of make LibCaral.rss is not actually rerunning CMAKE and therefore the install step is not aware of the fact that it has to copy the rss libraries over -> and so they are missing.

Maybe just try to delete all the Build/libcarla-client-build* folders and then run
make PythonAPI.rss
again.

berndgassmann, 2020-10-29:
At least the same procedure as you described works on my Ubuntu18.04 machine. (except that I had to replace 'make LibCarla.rss' by 'make LibCarla.client.rss' as the first target doesn't exist)
Then it makes use of the default python3 version installed which is python3.6.

Currently I try to run in the same carla folder:
make PythonAPI.rss  ARGS="--python-version=2.7,3.6,3.7"
and it should compile fine for all the three python versions, I've installed on my machine at once.
Maybe it's a python3.8 issue? But you said you also tested with other versions...

Hwansoo-Choi, 2020-10-29:
I've tried every methods you suggested, but I couldn't solve this problem, even tried with on Ubuntu18.04 machine and various python versions. (of course I tried with make LibCarla.client.rss, sorry for errata)

I was wondering if you could share your carla directory via git or google cloud
I know this method is very wasteful, but I cannot find any methods for solving the problem...
This is my mail address - hwansu97@gmail.com

I'm terribly sorry for asking too much...

direcf, 2020-10-29:
Thank you for your comments @berndgassmann

berndgassmann, 2020-10-29:
Hmm,
1.) the question is if the ad-rss-4.4.0 stuff was compiled correctly:
 find Build/ad-rss-4.4.0/install -name "*python*a"

gets me
`
Build/ad-rss-4.4.0/install/ad_map_access/lib/libad_map_access_python37.a
Build/ad-rss-4.4.0/install/ad_map_access/lib/libad_map_access_python27.a
Build/ad-rss-4.4.0/install/ad_map_access/lib/libad_map_access_python36.a
Build/ad-rss-4.4.0/install/ad_rss/lib/libad_rss_python37.a
Build/ad-rss-4.4.0/install/ad_rss/lib/libad_rss_python36.a
Build/ad-rss-4.4.0/install/ad_rss/lib/libad_rss_python27.a
Build/ad-rss-4.4.0/install/ad_physics/lib/libad_physics_python27.a
Build/ad-rss-4.4.0/install/ad_physics/lib/libad_physics_python36.a
Build/ad-rss-4.4.0/install/ad_physics/lib/libad_physics_python37.a
Build/ad-rss-4.4.0/install/ad_rss_map_integration/lib/libad_rss_map_integration_python27.a
Build/ad-rss-4.4.0/install/ad_rss_map_integration/lib/libad_rss_map_integration_python37.a
Build/ad-rss-4.4.0/install/ad_rss_map_integration/lib/libad_rss_map_integration_python36.a
`

after build all the three variants, for sure.
You then should see at least the respective python38 variants of your files.
As a quick fix you could just copy those then to your:
/home/hwansoo/carla/PythonAPI/carla/dependencies/lib/

But as it sounds to me the compilation of ad-rss-lib components could have went wrong already on your side.

find Build/ad-rss-4.4.0 -name "build-*" -type d

gives me:
`
Build/ad-rss-4.4.0/build-python3
Build/ad-rss-4.4.0/build-python3.7
Build/ad-rss-4.4.0/build-python3.6
Build/ad-rss-4.4.0/build-python2.7
`
the build-python3 is from the first build without ARGS (then python3 is used by default and the version of python that you get via python3

the others then are from the built with ARGS and detailed specified versions.

maybe you installed python3.8 and replaced the system python3 (which is 3.5 in Ubunut16.04) after doing:
pip3 install --user pygccxml pyplusplus

because then the respective pip3 install would have installed the packages for python3.5 and they could miss for your python3.8 installation?

you could also delete the Build/ad-rss-4.4.0/build* folders and trigger again a build of it via
make PythonAPI.rss

and then observe if there are error output on the build of ad-rss-4.4.0....

I'm sure we will find the root cause, so no problem.

Bernd.

berndgassmann, 2020-10-29:
Found an Ubunut16.04 machine to play around.

Found one possible issue:
the python binding generators for ad-rss-lib and its dependencies calls 'python' to perform the code generation.
pip3 install --user pygccxml pyplusplus
installs it for python3, but if the default python version in the system is different these packages are not found.

Within the build process it has to be there for the default python version (in my case python2.7); therefore a 
pip install --user pygccxml pyplusplus

made the python binding code generation work.

If your default python is 3.8 you might have to call
pip3.8 install --user pygccxml pyplusplus

to get the components installed for your default python version.

berndgassmann, 2020-11-02:
Hi,
I've tried to create a fix for the issue:
Make use of the python version to be used for the build also for the internal call to generate code with pyplusplus.
See:
https://github.com/intel/ad-rss-lib/pull/90
and
https://github.com/carla-simulator/map/pull/32

If these are merged, I'll create also a pull request for this CARLA repo to make use of that, in the hope it fixes your issue.

In the meantime, maybe you could copy the file
https://github.com/carla-simulator/map/pull/32/files#diff-60ea9a95a4f10c4f914e30682978976f6bdf383e642b465c97cc1a5b91c6b0b1
to the Build/ad-rss-4.4.0/src/map/cmake and Build/ad-rss-4.4.0/src/ad-rss-lib/cmake folders and see if that fixes your issue.

To get the RSS part rebuild you just have to delete the Build/ad-rss-4.4.0/build* folders and trigger a build of it via
make PythonAPI.rss.

Hwansoo-Choi, 2020-11-03:
Hello.

I've succeeded make PythonAPI.rss with my clean version of Ubuntu 18.04.

As you said, the problem was the "ad-rss-4.4.0" is not compiled properly, because of the python version binding issue.
In my guess, installing different versions of python via anaconda made it worse.
I'll try the way you proposed and let you know the result.

Anyway, thank you very much.
I wouldn't have succeeded without your help.

Best regards.
Hwansoo

closed 2020-11-03 by Hwansoo-Choi