carla-simulator/carla #3496: WalkerBoneControl Controller doesn't exactly what documentation does suggest

frodozet, 2020-10-22:
with carla 0.9.9 I try to bring a Walker Actor to live using manual control structure WalkerControl and especially WalkerBoneControl. WalkerControl has one example under PythonAPI/manual_control.py which works. WalkerBoneControl has no example code and a small documentation of some python lines, which - when executed - are giving the following result:
(attached image gives us a walker standing on his hands in carla.Location(0,0,0) ; probably relativ to the ground truth of the Walker??? )
My questions are:
a/ how the bone transforms are calculated relativ to their parents and walker coordinates and how should that be done without knowing of any initial bone position of the given Walker/Actor?
b/ how will the Walker be 'moved'? Or should we switch between a WalkerControl apply_control() and a WalkerBoneControl call to apply_control()?
[image]

corkyw10, 2021-02-24:
@bernatx can you follow up on this please? @frodozet thanks for your patience, we will look into it and try to get a better example into the docs

bernatx, 2021-02-24:
Hi,

That is a limitation of current implementation. The actual way of doing that, is to manage the whole skeleton of bones, and not only one because you can not retrieve the position of one specific bone. So, you need to take control of the whole skeleton position, for each bone. We know that is not a perfect solution, so probably in the future we will address this limitation.

Schuck84, 2021-05-03:
Hello,
I have also observed the same problem.
The current example isn't working and there is no documentation in which coordinate system (transforms) the bones must be controlled.
Are the transforms in world or local coordinates?
Is it possible to read out the initial transforms somewhere or calculate them?
Maybe UE4 Editor skeleton + actual world transforms in Simulation?

bernatx, 2021-05-03:
Hi,

All what you can do with pedestrian bones so far is documented here:
https://carla.readthedocs.io/en/latest/tuto_G_control_walker_skeletons/#manually-control-walker-bones
Bones are always relatives to his parent, so no world coordinates, they are local relative to parent.

analog-cbarber, 2021-05-03:
Yes, but those instructions really aren't sufficient. It requires a lot of careful experimentation to figure out how to pose the figure using this API.  The API isn't really parent relative. If you modify a joint without ever modifying the child joints, then indeed the whole unit will move relative to the parent joint, but the coordinates are relative to a point at the figure's feet, and are not relative to the parent joint (at least this is what I have observed using 0.9.9 on linux).

I was able to build an easier to use API on top of this, but it took a fair amount of work.

The other strange thing I noticed was that if you apply any bone control transforms then the figure's appearance changes to one particular figure based on gender, so you end up having only one male, one female and one child option when you pose them.

Schuck84, 2021-05-03:
@analog-cbarber could you tell me where to start in the communication chain from PYTHON API to Unreal C++ API to implement further commands? 
Like read out the actual transforms from the bones?
I want to understand how are the function calls nested.
Could you help me regarding this?

analog-cbarber, 2021-05-03:
I didn't actually look at the C++ code and there does not appear to be a way to read out the current settings. I just experimented with the Python code by setting each joint until it looked like it was in the right place more or less.

But in order to create a usable API based on relative rotations of the joints, you need to compute the chain of transforms back to the root location. I personally used my own 3D transform code to do this. 

Unfortunately, I cannot easily share my code for this since it is spread out over multiple code repos.

Schuck84, 2021-05-21:
Do you know where I can find some more details about the skeleton in unreal editor. There are transforms and reference transforms. For what are the references? I thought calculating the joints from it root over the skeleton parts is enough. But after the hip joint the calculations are wrong and I saw there are also reference transforms for each joint. It would be great to use the coordinates from unreal on the Carla python API...

Schuck84, 2021-05-25:
Hi,
have found a solution to control the bones properly by using the current implementation.
I'm observing an issue in the semantic segmentation right now, the walker isn't annotated as usual. 
Please see attached screenshot.
Could you help me to fix this, or point me out what causes this issue?
[image]

Thanks and regards

mzheng27, 2021-06-04:
Hi @Schuck84, I am also working on modifying walkers' bones from the client side. Your results look great! Can you share some more details on how you achieved it? I am a little confused about what you mean by "read the bone transforms from unreal editor". Thank you! 

I am not sure about the segmentation issue you are having, but I once saw black objects for a newly created semantic segmentation tag because I didn't correctly recompile/rebuild the uproject after changing some c++ files.

bernatx, 2021-06-07:
Hi,
Can you explain which version and how you spawn the pedestrian?
For current development version they seems to work.
[image]

closed 2023-06-02 by stale[bot]

cucdengjunli, 2025-04-13:
mark