carla-simulator/carla #3497: Lanes ground truth

PTorr, 2020-10-22:
Is there a way to get the ground truth of the lanes like in KITTI?

wendersonj, 2020-10-23:
Check this:
[](https://carla.readthedocs.io/en/latest/python_api/#carla.Waypoint)
and
[](https://carla.readthedocs.io/en/latest/core_map/) section about 'Lanes'

PTorr, 2020-10-25:
Hi,
thanks for the answer.
I tried this, it gives a waypoint of the lane but not all the coordinates.
Did you succeed to get the coordinates for the full lane?
Torr.

wendersonj, 2020-10-25:
Sorry, can't help anymore... But I'm trying to use this just now. If a succeed, I'l share with u.

wendersonj, 2020-10-26:
@PTorr , have you explored the sensor [Lane Invasion ](https://carla.readthedocs.io/en/latest/ref_sensors/#lane-invasion-detector) ?

PTorr, 2020-10-27:
I look into it.

wendersonj, 2020-10-28:
hi, @PTorr. I made some progress and I'm currently using this solution:
https://github.com/carla-simulator/carla/issues/1254
you can try retrieve the lanes (also left and right lanes) using waypoints and some arrangements in your code... you can also use OpenDrive API informations to retrieve something more specific.

PTorr, 2020-10-29:
Thanks @wendersonj I think #1254 its the right direction.

corkyw10, 2021-02-24:
@PTorr do you still need help with this issue?

PTorr, 2021-02-25:
@corkyw10 Hi,
I wasn't able to solve it for now.

corkyw10, 2021-03-05:
@doterop would you be able to have a look at this please?

doterop, 2021-03-29:
Hi @PyTorr ,

Can you explain more in-depth about what exactly you need?

PyTorr, 2021-03-29:
I wanted lanes segmentation.

Axel1092, 2021-04-13:
Hi @PyTorr, I don't know for sure what you mean but perhaps you need the semantic segmentation sensor that gives you the tags in an image including lanes and other items in the scene.

closed 2023-06-02 by stale[bot]