carla-simulator/carla #3499: Calibration matrix from Carla´s RGB camera

unzuetaunai, 2020-10-22:
Hello,

I am working with Carla's RGB camera and processing the data with Matlab. But, in order to calibrate the RGB camera in Matlab I need to complete the intrinsic matrix and the distortion matrix.
For the intrinsic matrix or calibration matrix I would need to obtain the focal lengths (fx,fy) of the camera in the x and y axis (parameters that are measured in pixels) and the displacement of the centre (cx,cy) of the image in x and y axis.
In the distortion matrix, both the radial and the tangential distortions appear, so for both cases, it would be necessary to have 2 of the distortion coefficients (K1,k2 for the radial distortion and p1,p2 for the tangential distortion) and the coordinates of the distorted pixel in x and y (xd,yd for the radial distortion and xd,yd for the tangential distortion)

The data regarding Carla's RGB camera can be found in the following link:
https://carla.readthedocs.io/en/0.9.9/ref_sensors/#rgb-camera

Then, is anyone able to tell me how I can obtain the above mentioned data from the camera datasheet of the previous link?

Thank you very much in advance.

GabiShahmayster, 2020-10-22:
Hi,
As far as I remember, any lens distortion is achieved by setting 'enable_postprocess_effects' to True, i.e there is no distortion by default.
All the other intrinsic parameters are just those of an ideal camera, that is (cx, cy) are the middle of the image and fx=fy=f where f can be calculated from the hfov and image width and there is no skew.
Hope that helps,
Gabi

unzuetaunai, 2020-10-26:
Thank you very much for your contribution.

unzuetaunai, 2020-10-27:
Hi Gabi,

I'm processing the images with the data you told me about, that is, with the null distortion matrix (parameters that I don't know if they are right or should be different from zero). But the data I'm getting is wrong or skips errors. Then, you could tell me if the data that appears in the section "Camera lens distortion attributes" has any meaning for the distortion matrix? I have read by some discussion that the values that appear in that section could be K1 (lens_k) or K2 (lens_kcube) which would be the values for the radial distortion, but for the tangential distortion I have not found anything about it. Although I am not sure about it, so if you knew or someone knew about it, it would be very helpful.

Thank you very much in advance.

GabiShahmayster, 2020-10-27:
Hi,
Are you sure that you are actually seeing distortion, rather than facing a different issue?
Do straight lines look curved in your images, for example?
Gabi

corkyw10, 2021-02-24:
@DSantosO can you follow up on this please?

DSantosO, 2021-02-25:
Hello @indartsu,

If you have not included distortion, the camera matrix should be just the one computed with fx,fy,cx,cy. Be careful with the frame transformation between Unreal's XYZ and the camera one that can be tricky. You can check the `lidar_to_camera.py` script in which this matrix is used to project points from 3D space to the camera image.

closed 2021-07-21 by stale[bot]