carla-simulator/carla #3502: Cannot run successful the setup script from carlaviz

Emiliyan192, 2020-10-23:
I'm trying to build the carlaviz plugin and a problem arises that I don't know how to handle. 
From the building instructions ( https://github.com/carla-simulator/carlaviz/blob/master/docs/build.md ) I have installed everything and a problem arise in the last step 3) of the "Step by Step " part by trying to run the setup script "./setup/setup.sh" . I've attached the whole output from the terminal windows. I have no idea what is the problem but I saved the whole output from the terminal window. I hope this will help further. 

Thank you in advance !
[output.txt](https://github.com/carla-simulator/carla/files/5428418/output.txt)

corkyw10, 2021-02-24:
@Emiliyan192 Thanks for your patience. Looks like the build instructions link doesn't exist anymore, are you still experiencing this issue?

@bernatx could you follow up please?

bernatx, 2021-02-24:
Hi,

The new link is here: https://github.com/wx9698/xviz
We have removed our fork and now the author is maintaining the plugin.

It seems that the problem is that the protobuf header is not found
`/home/emiliyan/carlaviz/backend/third_party/xviz/include/xviz/proto/options.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`
So, you need to check that part.

I hope it helps

closed 2021-07-21 by stale[bot]