carla-simulator/carla #3503: How to make the ground-truth of the bounding box of the point cloud？

mmmmaomao, 2020-10-23:
Hi, I am a student who is new to computer science. Recently, my teacher gave me a task: Use Carla to make a dataset similar to Kitty 3D target detection. Using the official rosbridge, I can collect lidar data, image data and OBJECT files. So, how do I make the bounding box tags of vehicles and pedestrians in the point cloud? I am grateful for your help!

ivannson, 2020-11-02:
You can use the semantic lidar sensor to get the ID for each point, then use something like DBSCAN (density-based clustering) to cluster all of the points with the same labels together. You might have some issues if the cars and pedestrians are very close together, as it might put two cars in one cluster, but tweak the hyperparameters to see how it works first. You might even want to use different parameters for different classes, e.g. separate clustering processes for pedestrians and cars. You're just making a dataset so I doubt the computational speed matters much. Then for each cluster get max and min coordinates to construct your bounding boxes.

Hope that helps!

mmmmaomao, 2020-11-04:
> You can use the semantic lidar sensor to get the ID for each point, then use something like DBSCAN (density-based clustering) to cluster all of the points with the same labels together. You might [...]

Thanks for your help!

closed 2021-02-24 by corkyw10