carla-simulator/carla #3505: Inconsistent LIDAR measurements in fixed-timestep synchronous mode

lynnpepin, 2020-10-23:
| | |
| - | - |
| OS | Ubuntu 18.04 |
| Carla | 0.9.10 |
| Python | 3.7.5 |

Hi all, I'm having an issue getting consistent measurements from LIDAR.

My goal is to get a predictable, consistently-sized LIDAR measurement per simulated frame. So, each frame should represent one 360-degree rotation of the sensor, and have the same number of points between each frame.

In the example I have here, I'm expecting to get captures of 16000-byte sized measurements, i.e. 1000-point captures, at each frame, but instead I get variable amounts over time, leveling off to around 504 after about 2 seconds (but this varies over different runs.)

The pertinent facts of the setup are described as follows:

1. Run CARLA:
    * Synchronous mode
    * Fixed timestep at 0.05s
    * FPS = 20
2. Set up LIDAR blueprint attributes follows:
    * Set all dropoff attributes to 0.0, so no points are dropped.
    * Set `rotation_frequency`to `20`
    * Set channels to 1, and points-per-second to 20000.
        * So, we should expect 1000 points captured in a single frame.
 3. Attach a listener function to the LIDAR, that simply dumps the raw_data into a Queue.

Plotting the results over time, we get these results:
[image]

Here, we see the number of points varies over the first several timesteps, and then levels off to around 8000 points (staying consistent for the rest of the simulation.) Over different runs, this stable level varies, producing around 503 or 504 points.

We should expect a flat line, at 16000 points, that stays consistent between runs. Does anybody know if this is a bug, or some form of user error? Please let me know! I appreciate CARLA greatly. 

If this is a bug without an existing fix, I'd also appreciate some pointers on where to start with the source code, I may be able to help fix it. I couldn't find anyone else with this issue on the GitHub, and I imagine my use-case must be common, so I'm hoping it's just a user error.

I also have a minimum example that showcases this issue:

```
###
# 0. Imports
###
import glob
import os
import random
import sys
import time
import numpy as np
from matplotlib import pyplot as plt

try:
    import queue
except ImportError:
    import Queue as queue

try:
    os_version = 'win-amd64' if os.name == 'nt' else 'linux-x86_64'
    sys.path.append(glob.glob(f'/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.10-py3.7-{os_version}.egg')[0])
except IndexError:
    pass

import carla

# Before run, start the Carla server as
# /opt/carla-simulator/CarlaUE4.sh --fps=20

###
# 1. Start CARLA server
###

actor_list = []
client = carla.Client('localhost', 2000)
client.set_timeout(2.0)
world = client.get_world()

settings = world.get_settings()
settings.fixed_delta_seconds = 0.05
settings.synchronous_mode = True
world.apply_settings(settings)

blueprint_library = world.get_blueprint_library()
bp = random.choice(blueprint_library.filter('vehicle'))

vehicle = world.spawn_actor(bp, carla.Transform(carla.Location(x=-50.0, y=1.0, z=0.5)))
actor_list.append(vehicle)
world.tick()

###
# 2. Set up lidar
###
lidar_sensor_bp = blueprint_library.find('sensor.lidar.ray_cast')
# set dropoff to 0.0, so all points are recorded
lidar_sensor_bp.set_attribute("dropoff_intensity_limit", "0.0")
lidar_sensor_bp.set_attribute("dropoff_zero_intensity", "0.0")
lidar_sensor_bp.set_attribute("dropoff_general_rate", "0.0")
# set rotation frequency to 20, matching our FPS
lidar_sensor_bp.set_attribute("rotation_frequency", "20")
# 1 channel, 20000 points per second, means we should have 1000 points per capture
lidar_sensor_bp.set_attribute("channels", "1")
lidar_sensor_bp.set_attribute("points_per_second", "20000")

# pc_queue to hold pointcloud data
pc_queue  = queue.Queue()

def lidar_handler(point_cloud, pc_queue=pc_queue):
    pc_queue.put(point_cloud.raw_data)

lidar.listen(lidar_handler)

###
# 3. Start capturing pointcloud data
###

sizes = []

for ii in range(100):
    print(f"Step {ii}...")
    world.tick()
    raw_pc = pc_queue.get()
    sizes.append(len(raw_pc))

###
# 4. Plot the size of each pointcloud data
###

plt.plot(range(100), sizes)
plt.show()
# We should expect a horizontal line, at 16000
# (16000 bytes = 4000 floats = 1000 points)
```

lynnpepin, 2020-10-23:
I think the issue might be the following:

- The LIDAR does not report measurements outside its range, e.g. those that hit the sky.
- The vehicle spawns inside the ground.
- As it "bumps" out of the ground, the LIDAR moves with it, recording different amounts of points, until it balances it out.
- Small random pertubations in the physics cause it to be different between runs.

If this is a correct explanation, then I am wondering: Is it possible to set the LIDAR to fill in these points, e.g. with NaN or inf? (In this manner, a given point at a given index would always correspond to a given orientation relative to the vehicle. As an example, this would be useful for processing data for a neural-network.)

closed 2020-10-27 by lynnpepin

ivannson, 2020-11-02:
As @lynnpepin has said, you cant expect to have the same number of points, as Lidar will not return a value if it didn't hit anything (just like the real sensor). It is however possible to fill in those missing values by looking at the spherical coordinates of each point. In the spherical coordinate system, each point is described by phi (vertical angle), theta (horizontal angle) and r (distance to the point). Each combination of phi and theta is unique, so you could check whether a point is present by checking it's phi and theta values. You could pretty much have a template, where your phis and thetas are known, and you're just filling in the missing information for r, however I'm not sure if that's feasible in real time if you are just iterating through each point in the scan, but maybe you could look into using Numpy's mask. 

A similar approach is used when converting a Lidar scan into a depth image for Lidar object classification, where a mask is applied on the lidar scan, and the missing point are given a value of zero.