carla-simulator/carla #3507: Error while osm2odr map file conversion... Kindly suggest solution

LVJ047, 2020-10-26:
**Error detail:**
Warning: Cannot read local schema '../carla/data/xsd/types_file.xsd', will try website lookup.
Traceback (most recent call last):
  File "mapgeen.py", line 21, in <module>
    xodr_data = carla.Osm2Odr.convert(osm_data, settings)
RuntimeError: unable to open file 'https://sumo.dlr.de/xsd/types_file.xsd'
 In file 'built in type map'
 At line/column 1/0.
 The types could not be loaded from 'built in type map'.

**Queries:**

1) Why it is showing the warning "Cannot read local schema...." I checked it is not missing in the directed location. I believe the error probably is in the path definition to the "types_file.xsd". May I know if I can edit this location in the convert() definition,if yes how?

2)  It is unable to access the web location indicated what is the solution for this.

Thank you in advance for your assistance.

Asylbeck, 2020-11-23:
Hey try running it from the "PythonAPI/carla" folder, seems to work for me

corkyw10, 2021-02-24:
@LVJ047 Are you still experiencing this error?

@Axel1092 could you follow up please if required?

LVJ047, 2021-03-02:
@corkyw10 I could resolve it temporarily using Asylbeck's input.

Thanks

ar4development, 2021-03-10:
@LVJ047 if you execute your code from IDe like PyCharm you need to configure your run to have working directory at `PythonAPI/carla`

Axel1092, 2021-03-16:
As it has been already said, you need to run the scripts from the PythonAPI\{util,examples} folder in order for all files to be found.

closed 2021-03-29 by LVJ047