carla-simulator/carla #3508: [Testing] Raycast capabilities

Axel1092, 2020-10-26:
New functions have been added: `world.cast_ray(init, end)`, `world.project_point(pos, dir, dist)` and `world.ground_projection(pos, dist)`. These return `LabelledPoint` which is a position in space and a semantic tag. 
`cast_ray` return intersections between the line connecting init and end and the geometry of the scene.
`project_point` returns only the first intersection of a particular point and a specified direction.
`ground_projection`is a specialized version of `project_point`

joel-mb, 2020-11-02:
@Axel1092 works great!

closed 2021-01-26 by germanros1987

zizo1111, 2022-12-13:
Hi @Axel1092 @joel-mb @germanros1987 ,
do these methods follow the exact shape of the geometries?
I have tried out the methods and found that, when casting many points around an object the resulting shape is a simplified representation of how it looks like. I have noticed also that a point cloud produced by the Lidar provides more details about the geometries.
Is this really the case, or is it a problem on my side.
Thanks in advance.