carla-simulator/carla #3510: LiDAR Data acquisition

danielaDESM, 2020-10-26:
Hi! 

I am trying to understand a little bit more about the relationship between FPS, time_step, LiDAR rotation frequency and sensor tick. I have read the documentation and posts from other people that actually have helped me a lot. I just want to clarify the following:
If I set in my world settings: 
      settings.fixed_delta_seconds= 0.1 & synchronous mode (therefore FPS= 10)
And my LiDAR configuration:
        lidar_bp.set_attribute('points_per_second','280000') 
        lidar_bp.set_attribute('rotation_frequency','10')
      
And I am using the lidar.listen(lambda data: record(data)) to write in a PCD file the data of that measurement. At the end I should have 10 PCD files with 28,000 points each file per simulated second, right? But instead of that, I am having around 90-98 files per simulated second and I do not understand why. 
Another thing that I have noticed is that if I run through command line, either of the following commands:
CarlaUE4.exe -carla-port=2000 -quality-level={Epic} -windowed -ResX=500 -ResY=400 -fps=10
CarlaUE4.exe -carla-port=2000 -quality-level={Epic} -windowed -ResX=500 -ResY=400 -benchmark -fps=10
a warning appears in the terminal when I run my python script stating that I should define a fixed time step, because I am in synchronous mode, which tells me that the the -fps=10 is being ignored, right?
When I add the configuration within the script (as seen above) settings.fixed_delta_seconds= 0.1, the warning does not appear anymore but the speed in the visualization of the simulation does not change. 

Am I understanding wrongly the documentation?

Thanks in advance :)

corkyw10, 2021-02-24:
@danielaDESM Thanks for your patience. There's been changes to synchronous mode in the last release so Dani might be able to help you with updated info about this query

@DSantosO could you follow up please?

DSantosO, 2021-02-24:
Hello @danielaDESM,

As you are doing, the correct way of setting the simulator options is through the world settings. As far as I know, the benchmark and fps arguments of the server are not used. 
For the lidar topic, you should receive a 'file' per step, not per simulated seconds. So as dt=0.1, you get 10 files per simulated second. The number of points per second is much less than `points_per_second*dt` because those are the ones that the sensor try to compute, but most of the rays are not going to hit any object (for example the ones that point to the sky) and then we don't receive anything for that 'point'. 
The only thing to be aware of is that the callback of the sensor is triggered when the sensor data is received so they can be behind the world tick. In order to synchronize them, you can look our `sensor_synchronization.py` script in which we show how to do that.

danielaDESM, 2021-02-25:
Thanks a lot,
I did actually manage to understand it better. I do get the number of files I am supposed to according to the dt. As we wanted our simulation to run in more FPS and that does not necessarily mean that we want the LiDAR to acquire the data at the same rate, we keep the clouds in a queue and fill a whole rotation accordingly.  The sensor_synchronization.py helped a lot.  
Thanks in advance for the answer :)

closed 2021-03-10 by DSantosO

Gatsby23, 2022-04-08:
> Hello @danielaDESM,
>
> As you are doing, the correct way of setting the simulator options is through the world settings. As far as I know, the benchmark and fps arguments of the server are not used. [...]

I'm sorry, I have the same problem too.
Could you please give a more clearly advice about how to set different sensors' frequency more clearly？