carla-simulator/carla #3512: Lane Invasion when reversing doesn't work ?

wendersonj, 2020-10-26:
Hi !
Lane Invasion Sensor doesn't recognize lane mark when going **reverse**...  Possible bug ? 
Standard `manual_control.py` from Carla 0.9.9.4

[image]

wendersonj, 2020-10-26:
Note: it did works when half car when over the lane.

thytless, 2020-10-27:
Standard `manual_control.py` from Carla 0.9.9.4 as well, looks fine on Town06...
Maybe try a little more distance?

wendersonj, 2020-10-28:
what do you means by a little more distance?

wendersonj, 2020-10-28:
when driving with low speed and some angles of the car, the sensor doesn't "see" the lane...

corkyw10, 2021-02-24:
@Axel1092 could you have a look at this please?

Axel1092, 2021-03-03:
Hi all, I have checked the issue and the problem is that the lane invasion sensor relies on the OpenDRIVE data to detect the lane markings but these are not exported within junctions. The example you are showing a situation which is labelled as junction in OpenDRIVE (even if only the opposite side of the road has connections with other roads.