carla-simulator/carla #3513: Is it possible to simulate a mobile robot(for outdoor) under CARLA?

jediofgever, 2020-10-27:
Question rather than issue. 

Hello, 

I really like visual capabilities of CARLA, I am aiming to simulate a mobile robot on outdoor envoirment. 

I am well aware that CARLA's primary target is Vehicle simulation. However I wonder whether it will be possible to simulate lets say a omni directional mobile robot. ?

Did anyone tried this before ? 

thank you for your time

jimmyw404, 2020-10-28:
It's possible, yes, but if you're not using the roadway network and the many on-road features that CARLA provides, you're not really making much usage of it. If you don't have a mechanism to do vehicle-terrain-interaction modeling in off-road situations or a way to model the particular configuration of your robot, you're missing out on a lot that makes a robot simulator useful.

Depending on what you're doing you might want to take a look at Webots, VREP/Coppelia or Gazebo.

zhiranyan, 2021-03-18:
Hi, I also want to simulate a omni directional mobile robot in Carla. Others simulators do not fit for my task. So do you have a solution? Any tipps, I appreciate!

closed 2023-06-02 by stale[bot]