carla-simulator/carla #3514: include xdg-user-dirs xdg-utils packages in the Docker image

qhaas, 2020-10-27:
A number of users see the following (seemingly) harmless warning when using the docker image and become concerned or 
 worse, it becomes a red herring while troubleshooting unrelated issues:
```
sh: 1: xdg-user-dir: not found
```

Recommend deploying the `xdg-user-dirs xdg-utils` packages [in the Docker image](https://github.com/carla-simulator/carla/blob/0.9.10.1/Util/Docker/Release.Dockerfile#L15) to prevent this warning from appearing and reduce the number of issues posted here and/or time spent by users who see this as the last console entry and assume it has something to do with whatever issue they are experiencing.

corkyw10, 2021-02-24:
@bernatx could you have a look at this and provide your opinion?

qhaas, 2021-02-24:
A less invasive approach would be to find a way to suppress said warning, which I have not yet discovered.  Adding said additional packages might have unintended consequences beyond a larger image.

qhaas, 2021-07-07:
Since this continues to be a red herring with the community and is likely an UE nuance, [I asked over in UE Answers](https://answers.unrealengine.com/questions/1039260/suppress-or-fix-xdg-user-dir-not-found.html) how to suppress it.

rzeydelis, 2022-01-15:
`xhost local:root` fixed this issue for me.

^^ Type this command first and then try running Carla again:
```
sudo docker run   -e SDL_VIDEODRIVER=x11   -e DISPLAY=$DISPLAY  -v /tmp/.X11-unix:/tmp/.X11-unix   -p 2000-2002:2000-2002   -it   --gpus all   carlasim/carla:0.9.10 ./CarlaUE4.sh -opengl
```

closed 2023-06-02 by stale[bot]

muety, 2024-02-20:
This is still not fixed, is it?

qhaas, 2024-12-25:
Looks like the devs fixed:  https://github.com/carla-simulator/carla/commit/ada75f920642e18cace0a9f85ecf9d4077ddb531#diff-083cbe5141591350f2430e8feb2390a1114e59ba5d8c9728e596d4c5f176fc00R5