carla-simulator/carla #3516: RGB Camera artifacts issue Carla 0.9.10.1

HoucemF, 2020-10-27:
Hi Carla Team,
I am using Carla 0.9.10.1 running on ubuntu 18.04 using python 3.7.9. I am collecting data from the RGB camera and from the semantic segmentation camera. I am using the Low Quality preset for rendering and rendering using Vulcan. The issue is the appearance of artifacts in the RGB images. These artifacts were not visible using the spectator camera.

RGB Image
[image]

Semantic Segmentation Image
[image]

qhaas, 2020-10-27:
Reminds me of Issue ##3377

HoucemF, 2020-10-29:
In that issue the artifacts appear in spectator mode while I only found them when using the camera
[image]
 
This image was captured running with the -opengl flag.

corkyw10, 2021-02-24:
@DSantosO could you have a look at this please?

DSantosO, 2021-02-25:
Hello @HoucemF,

Can you still see this problem in the 0.9.11 release?

qhaas, 2021-02-25:
We are still getting the artifacts with OpenGL + official/pre-built CARLA 0.9.11 (UE4.24), they are tiny black '+' if you zoom in.  They are most evident in 'HardRainSunset' weather.  They don't exist when we use Vulkan+UE4.25, but since we are x-server-less, we had to upgrade to UE 4.25 to get headless/xless vulkan support (see Issue #3800)
[image]

Esquilli, 2021-03-11:
Bumping this. I'm still having this issue in the latest version of CARLA. It get worse when visualizing the data from the sensors (e.g depth, semantic, etc.). More info [here](https://github.com/carla-simulator/carla/issues/3865). 

I can upload more pictures if someone want them.

HoucemF, 2021-03-23:
Sorry for the late response, I am still working with 0.9.10.1 and I have discovered another more serious issue with the camera. At some points, when the camera is collecting images it will turn blue and sometimes completely pitch black for a number of frames. 
[image]
[image]
The occurrence of this issue is for 1 out of 10 simulations and if it happens once there is a high probability that it happens again in the same simulation.

closed 2023-06-02 by stale[bot]