carla-simulator/carla #3517: [Testing] PR #3515, Adding load control on sublevels

doterop, 2020-10-27:
This issue is intended to track PR #3515 .

Remember that this feature only works on maps created with sub-levelling, maps with the `_opt` suffix.

Need to test:
- Only the selected labels are loaded.
- Only the selected labels are unloaded.
- If a layer or layers are selected when a new map is loaded by `load_world`, only those layers are loaded.
- The new maps are not affected by this feature, ie: calling any load/unload function does not have any effect.

The manual_control has been updated to show this feature and small modifications on the script are enough to test this feature. Although, feel free to experiment.

marcgpuig, 2020-11-02:
So far the examples work great and the API doesn't require any change to the code to keep working as usual

XGodina, 2020-11-02:
I've tested every Opt map with manual control on the windows and it works as expected

closed 2021-01-26 by germanros1987

pavanteja295, 2021-07-04:
Hello , 
I am using carla release 0.9.11 with Town01_opt map trying to load and unload map layers. I am then saving these images using sensor synchronization. However, the map layers do not work as expected. An example scenario is the following.

`world = client.load_world(args.map , carla.MapLayer.Ground)`

Following is an example of saved image. 
[image]

I also tried specifically unloading layers like:
`    world.unload_map_layer(carla.MapLayer.All)
    world.load_map_layer(carla.MapLayer.Ground)`

This resulted in the following image:
[image]

It would be great if someone let me know what is wrong with my code? Do you recommend compiling carla from source?

cjychenjiayi, 2024-12-06:
Hello, I use the release version of carla, 0.9.13, and when i try to remove the Props in Town01_Opt, it failed.
Removing ParkingVehicle work in Town01_Opt, removing Props work in Town10HD_Opt
I dont know why