carla-simulator/carla #3521: Building Carla in Docker: make package cook failed

gismo07, 2020-10-28:
I'm following the official guide [Building Carla in a Docker](https://github.com/carla-simulator/carla/tree/dev/Util/Docker) to build the Unreal Engine and Carla Docker containers in order to create packages for own maps.

What we did:
1. we cloned version 0.9.10.1:
    ```shell
    git clone https://github.com/carla-simulator/carla.git
    cd carla
    git checkout 0.9.10.1
    ```
2. build the Unreal Engine image:
    ```shell
    cd Util/Docker
    pip3 install ue4-docker
    ue4-docker setup
    ue4-docker build 4.24.3 --no-engine --no-minimal
    ```
3. build the carla image:
    ```shell
    docker build -t carla-prerequisites -f Prerequisites.Dockerfile .
    docker build -t carla -f Carla.Dockerfile .
    ```

However, the last Carla.Dockerfile command, `make package` fails with the error message:
```txt
Log.WriteException: ==============================================================================
Log.WriteException: ERROR: Cook failed.
Log.WriteException:        (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
Log.WriteException: 
Log.WriteException: CommandletException: Editor terminated with exit code 137 (signal 9) while running Cook for /home/ue4/carla/Unreal/CarlaUE4/CarlaUE4.uproject; see log /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.10.27-23.22.36.txt
Log.WriteException:   at AutomationTool.CommandUtils.RunCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String Commandlet, System.String Parameters, System.String& DestLogFile) [0x0076d] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at AutomationTool.CommandUtils.RunCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String Commandlet, System.String Parameters) [0x00001] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at AutomationTool.CommandUtils.CookCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String[] Maps, System.String[] Dirs, System.String InternationalizationPreset, System.String[] CulturesToCook, System.String TargetPlatform, System.String Parameters) [0x0013e] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at Project.Cook (AutomationTool.ProjectParams Params) [0x007ba] in <712eb0506aad4907b6a922f8f9ab5dba>:0 Wrapped by AutomationException: Cook failed.
Log.WriteException:   at Project.Cook (AutomationTool.ProjectParams Params) [0x007f7] in <712eb0506aad4907b6a922f8f9ab5dba>:0 
Log.WriteException:   at BuildCookRun.DoBuildCookRun (AutomationTool.ProjectParams Params) [0x00074] in <712eb0506aad4907b6a922f8f9ab5dba>:0 
Log.WriteException:   at BuildCookRun.ExecuteBuild () [0x00040] in <712eb0506aad4907b6a922f8f9ab5dba>:0 
Log.WriteException:   at AutomationTool.BuildCommand.Execute () [0x00001] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at AutomationTool.Automation.Execute (System.Collections.Generic.List`1[T] CommandsToExecute, System.Collections.Generic.Dictionary`2[TKey,TValue] Commands) [0x0007c] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at AutomationTool.Automation.Process (System.String[] Arguments, UnrealBuildTool.StartupTraceListener StartupListener) [0x00282] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at AutomationTool.Program.MainProc (System.String[] Arguments, UnrealBuildTool.StartupTraceListener StartupListener) [0x00001] in <5c1228a498974a8686b196c864089aae>:0 
Log.WriteException:   at AutomationTool.Program+<>c__DisplayClass1_0.<Main>b__2 () [0x00000] in <5c1228a498974a8686b196c864089aae>:0 
Log.WriteException:   at AutomationTool.InternalUtils.RunSingleInstance (System.Func`1[TResult] Main) [0x000a0] in <ca38fb0390fe4bd797c39ab78cd1d27a>:0 
Log.WriteException:   at AutomationTool.Program.Main (System.String[] Arguments) [0x0020a] in <5c1228a498974a8686b196c864089aae>:0
Log.WriteException: ==============================================================================
Program.Main: AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
```

Now the question is how to resolve this issue? Is Unreal Engine `4.24.3` sufficent for carla `0.9.10.1`?

Or should we pass the git tag to all carla related docker commands like:
```shell
docker build -t carla -f Carla.Dockerfile . --build-arg GIT_BRANCH=0.9.10.1
```

gismo07, 2020-10-28:
I'm also adding the full log files
[Cook-2020.10.27-23.22.36.txt](https://github.com/carla-simulator/carla/files/5451481/Cook-2020.10.27-23.22.36.txt)
[Log.txt](https://github.com/carla-simulator/carla/files/5451482/Log.txt)
[UBT-CarlaUE4-Linux-Shipping.txt](https://github.com/carla-simulator/carla/files/5451484/UBT-CarlaUE4-Linux-Shipping.txt)
[UBT-UnrealPak-Linux-Development.txt](https://github.com/carla-simulator/carla/files/5451485/UBT-UnrealPak-Linux-Development.txt)

corkyw10, 2021-02-25:
@gismo07 Thanks for your patience, are you still experiencing this issue?

@bernatx could you follow up on this please?

closed 2021-07-21 by stale[bot]