carla-simulator/carla #3523: issues transforming openstreetmap to opendrive format

ThierryDeruyttere, 2020-10-28:
So I follow the instructions from the documentation on how to convert the openstreetmap format to open drive.
However, I get an error saying: `RuntimeError: No nodes loaded.` 

The code that I use is: `    f = open(osm, 'r')
    osm_data = f.read()
    f.close()
    # Define the desired settings. In this case, default values.
    settings = carla.Osm2OdrSettings()
    # Convert to .xodr
    print(len(osm_data), settings)
    xodr_data = carla.Osm2Odr.convert(osm_data, settings)

    # save opendrive file
    f = open(osm.split("/")[-1].replace("osm", "xodr"), 'w')
    f.write(xodr_data)
    f.close()
`
File to recreate the issue:
[sing-one.zip](https://github.com/carla-simulator/carla/files/5454119/sing-one.zip)

corkyw10, 2021-02-25:
@Axel1092 could you have a look at this please?

Axel1092, 2021-03-03:
Hi @ThierryDeruyttere, sorry for the late response, apparently your OSM data does not contain the type of the roads it is exporting (highway, railway, waterway, cycleway, busway, route, sidewalk, highspeed or usage) which is necessary to properly generate the representation of the road. How are you generating this data?

ThierryDeruyttere, 2021-03-03:
I was using https://github.com/gdlg/nuscenes-to-osm-exporter to generate the file I believe.

Axel1092, 2021-03-15:
After seeing the code you would have to add the road type for our converter to work. Something like changing in `nuscenes_map_to_osm_exporter.py` line 110 with `export_polygon(output_file, numap, 'drivable_area', record, '<tag k="highway" v="residential"/>')`. The issue with not having the tag `"highway"` is that we cannot know if the way is defining a road or a shape of something different.

driverych, 2021-06-08:
Have you found a valid solution to this problem? I believe I have fallen into a similar circumstance as you. I have been trying to convert a .osm file in the Lanelet2 structure to a .xodr file, which could be imported by the Driving Scenario Designer of Matlab. 

The Carla version is 0.9.11, the Ubuntu 18.04, operated with Intel i7-8700K and Nvidia GTX1070. 

[location1.zip](https://github.com/carla-simulator/carla/files/6614104/location1.zip)
The python code is as follows:

import carla
f = open("location1.osm", 'r') # Windows will need to encode the file in UTF-8. Read the note below.
osm_data = f.read()
f.close()

settings = carla.Osm2OdrSettings()
xodr_data = carla.Osm2Odr.convert(osm_data, settings)

f = open("location.xodr", 'w')
f.write(xodr_data)
f.close()

And these codes are performed in the Examples folder of PythonAPI. The Carla is installed both through Linux build and a recommended quick start installation.

Could you help with this problem? The .osm file is attached. Any suggestions will be helpful.

closed 2023-06-02 by stale[bot]

muety, 2023-08-09:
I tried to convert Lanelet maps in OSM format to OpenDRIVE and ran into the same issue as well. Any ideas?

devanshanker, 2023-10-20:
@driverych how did you resolve this issue in the end?