carla-simulator/carla #3528: include bounding box extent to distance calculation in is_vehicle_hazard()

mselli, 2020-10-31:
the function [is_vehicle_hazard()](https://github.com/carla-simulator/carla/blob/7a14a531075eecf4ff5ea88a31cff230b0481cfb/PythonAPI/carla/agents/navigation/agent.py#L189) within the Agent does not take into account the actors bounding box's extent for calculating whether another vehicle is within the defined proximity_vehicle_threshold or not. 
In the call to [is_within_distance_ahead()](https://github.com/carla-simulator/carla/blob/7a14a531075eecf4ff5ea88a31cff230b0481cfb/PythonAPI/carla/agents/navigation/agent.py#L213) only the actors' transformations are passed, and therefore, a distance calculation is made with respect to the vehicles' center point, not with respect to the extent of each vehicle's bounding box. 

So, for example, in a car following situation where both vehicles are driving on the same lane 2 meters away from each other with proximity_vehicle_threshold = 2 meters, the method will never detect a hazard situation when the headway distance is less than 2 meters because the distance calculated by is_within_distance_ahead() has always a positive offset that is given by  vehicle1_length/2 + vehicle2_length/2.

sagar-g-v, 2020-11-02:
Hi @mselli ,

I have expanded is_vehicle_hazard() function considering tv bounding box also relative velocity.

```python3
       def _is_vehicle_hazard(self, vehicle_list):
        """

        :param vehicle_list: list of potential obstacle to check
        :return: a tuple given by (bool_flag, vehicle), where
                 - bool_flag is True if there is a vehicle ahead blocking us
                   and False otherwise
                 - vehicle is the blocker object itself
        """
        ego_vehicle_location = self._vehicle.get_location()
        ego_vehicle_waypoint = self._map.get_waypoint(ego_vehicle_location)

        longitudinal_acc = self._vehicle.get_acceleration().x
        ego_velocity = self._vehicle.get_velocity().x
        response_time = 0.8

        safe_distance = self._proximity_vehicle_threshold

        for target_vehicle in vehicle_list:

            # recalculate safe_distance
            if longitudinal_acc - target_vehicle.get_acceleration().x < 0.0:
                safe_distance = response_time * ego_velocity
                if safe_distance < self._proximity_vehicle_threshold:
                    safe_distance = self._proximity_vehicle_threshold
            # do not account for the ego vehicle
            if target_vehicle.id == self._vehicle.id:
                continue

            tv_vertices = target_vehicle.bounding_box.get_world_vertices(
                target_vehicle.get_transform())
            tv_vertices = [tv_vertices[0], tv_vertices[2], tv_vertices[4], tv_vertices[6]]
            tv_vertex_waypoints = [self._map.get_waypoint(vertex) for vertex in tv_vertices]

            # check if object vertices falls on ego lane
            extension_falls_on_ego_lane = False
            for tv_vertex_waypoint in tv_vertex_waypoints:
                if tv_vertex_waypoint.lane_id == ego_vehicle_waypoint.lane_id:
                    extension_falls_on_ego_lane = True

            # if the object is not in our lane it's not an obstacle
            target_vehicle_waypoint = self._map.get_waypoint(target_vehicle.get_location())
            if (target_vehicle_waypoint.road_id == ego_vehicle_waypoint.road_id and
                target_vehicle_waypoint.lane_id == ego_vehicle_waypoint.lane_id) or \
                    extension_falls_on_ego_lane:
                delta_p = self._vehicle.bounding_box.extent.x + target_vehicle.bounding_box.extent.x
                if is_within_distance_ahead(target_vehicle.get_transform(),
                                            self._vehicle.get_transform(),
                                            safe_distance + delta_p):
                    return True, target_vehicle

        return False, None
```

Hope It will be helpful for you.

corkyw10, 2021-02-25:
@jackbart94 could you follow up on this please?

GyChou, 2021-04-03:
> Hi @mselli ,
> 
> I have expanded is_vehicle_hazard() function considering tv bounding box also relative velocity.
> 
> ```python
>        def _is_vehicle_hazard(self, vehicle_list):
>         """
> 
>         :param vehicle_list: list of potential obstacle to check
>         :return: a tuple given by (bool_flag, vehicle), where
>                  - bool_flag is True if there is a vehicle ahead blocking us
>                    and False otherwise
>                  - vehicle is the blocker object itself
>         """
>         ego_vehicle_location = self._vehicle.get_location()
>         ego_vehicle_waypoint = self._map.get_waypoint(ego_vehicle_location)
> 
>         longitudinal_acc = self._vehicle.get_acceleration().x
>         ego_velocity = self._vehicle.get_velocity().x
>         response_time = 0.8
> 
>         safe_distance = self._proximity_vehicle_threshold
> 
>         for target_vehicle in vehicle_list:
> 
>             # recalculate safe_distance
>             if longitudinal_acc - target_vehicle.get_acceleration().x < 0.0:
>                 safe_distance = response_time * ego_velocity
>                 if safe_distance < self._proximity_vehicle_threshold:
>                     safe_distance = self._proximity_vehicle_threshold
>             # do not account for the ego vehicle
>             if target_vehicle.id == self._vehicle.id:
>                 continue
> 
>             tv_vertices = target_vehicle.bounding_box.get_world_vertices(
>                 target_vehicle.get_transform())
>             tv_vertices = [tv_vertices[0], tv_vertices[2], tv_vertices[4], tv_vertices[6]]
>             tv_vertex_waypoints = [self._map.get_waypoint(vertex) for vertex in tv_vertices]
> 
>             # check if object vertices falls on ego lane
>             extension_falls_on_ego_lane = False
>             for tv_vertex_waypoint in tv_vertex_waypoints:
>                 if tv_vertex_waypoint.lane_id == ego_vehicle_waypoint.lane_id:
>                     extension_falls_on_ego_lane = True
> 
>             # if the object is not in our lane it's not an obstacle
>             target_vehicle_waypoint = self._map.get_waypoint(target_vehicle.get_location())
>             if (target_vehicle_waypoint.road_id == ego_vehicle_waypoint.road_id and
>                 target_vehicle_waypoint.lane_id == ego_vehicle_waypoint.lane_id) or \
>                     extension_falls_on_ego_lane:
>                 delta_p = self._vehicle.bounding_box.extent.x + target_vehicle.bounding_box.extent.x
>                 if is_within_distance_ahead(target_vehicle.get_transform(),
>                                             self._vehicle.get_transform(),
>                                             safe_distance + delta_p):
>                     return True, target_vehicle
> 
>         return False, None
> ```
> 
> Hope It will be helpful for you.

It should be noted that `self._vehicle.get_velocity().x` is the velocity on the x-axis, not the longitudinal ego velocity

PR #3973 (2021-04-27): Improved agent vehicle detection
### Description

Improved the `is_vehicle_hazard` function at the Python API agents to also take into account both the ego's and the adversary's bounding boxes.

Fixes  #3528

### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.26

### Possible Drawbacks

None?

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3973)
closed 2023-06-02 by stale[bot]