carla-simulator/carla #3529: How can I change the player with keyboard

Fangwq, 2020-11-01:
I am new to the Carla. And I want to change the player with keyboard to observe other vehicles, but I do not know how to pick the player randomly. Does anybody know? Thank you very much.

XGodina, 2020-11-02:
Hi @Fangwq 
When you run the manual control, if you press "h" it shows you the help. But if you want to change the player you need press the Backspace key

Fangwq, 2020-11-03:
Thank you for your answers. The Backspace key only changes the style of the vehicle, and it is in the same position. But, I want to choose different vehicles at different position.

XGodina, 2020-11-03:
@Fangwq 
The backspace allows you to switch between motorcycles, bicycles, and cars. If you want to spawn at a particular point you must add these lines in manual_control script. Then always spawn to the same place you want.

spawn_points = world.get_map().get_spawn_points()
actor = world.spawn_actor(prop, spawn_points[0])

Fangwq, 2020-11-03:
OK, Thank you very much

closed 2020-11-03 by Fangwq