carla-simulator/carla #3530: Is there no very hard raining like CARLA 0.8 in CARLA 0.9?

Kitsunetic, 2020-11-01:
[image]

I'm finding hard raining like this. This picture was taken on CARLA 0.8.4.

Now I have to use version 0.9.9.4, and I have tested with weather parameters to make hard raining condition like this picture but I can't find it.

I have tested with all weather presets in `carla.WeatherParameters` and I manually set `precipitation` as large value but I never get weather like this picture.

`HardRainSunset` was similar with that but not that much raining.

Madecu, 2020-11-19:
Hi @Kitsunetic, we have changed the amount of rain particles with the new weather. Right now, to fully customize the rain, you have to go to Unreal find the PS_Rain and increase the maximum particles
[image]

You can also tweak the "color" of the raindrops to make them whiter and more visible as they were made before the update. 
[image]

I hope it helps

Kitsunetic, 2020-11-21:
Great! Thank you!!

I'll close the issue

closed 2020-11-21 by Kitsunetic

QAbot-zh, 2024-05-29:
> Hi @Kitsunetic, we have changed the amount of rain particles with the new weather. Right now, to fully customize the rain, you have to go to Unreal find the PS_Rain and increase the maximum [...]

How can I modify it? I open it on UE and it shows that the baked asset can not be modified. Is there another way to modify the parameters of PS_Rain?