carla-simulator/carla #3531: cordmup in localizationstage

hellojql, 2020-11-02:
#0  0x00007f6ba986d428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007f6ba986f02a in __GI_abort () at abort.c:89
#2  0x00007f6ba68d7dde in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f6ba68e3896 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f6ba68e3901 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f6ba68e3b55 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f6ba68da488 in std::__throw_out_of_range(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007f6b78a8c6ec in std::__detail::_Map_base<unsigned int, std::pair<unsigned int const, carla::traffic_manager::KinematicState>, std::allocator<std::pair<unsigned int const, carla::traffic_manager::KinematicState> >, std::__detail::_Select1st, std::equal_to<unsigned int>, std::hash<unsigned int>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true>, true>::at (this=<optimized out>, __k=<optimized out>) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/hashtable_policy.h:793
#8  std::unordered_map<unsigned int, carla::traffic_manager::KinematicState, std::hash<unsigned int>, std::equal_to<unsigned int>, std::allocator<std::pair<unsigned int const, carla::traffic_manager::KinematicState> > >::at (
    this=<optimized out>, __k=<optimized out>) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/unordered_map.h:994
#9  carla::traffic_manager::SimulationState::GetLocation (this=<optimized out>, actor_id=2332) at ../../LibCarla/source/carla/trafficmanager/SimulationState.cpp:46
#10 0x00007f6b78a7f18b in carla::traffic_manager::LocalizationStage::Update (this=0x46c7dd0, index=0) at ../../LibCarla/source/carla/trafficmanager/LocalizationStage.cpp:35
#11 0x00007f6b78a2598b in carla::traffic_manager::TrafficManagerLocal::Run (this=0x46c7630) at ../../LibCarla/source/carla/trafficmanager/TrafficManagerLocal.cpp:176
#12 0x00007f6ba690eb10 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007f6ba9c096ba in start_thread (arg=0x7f6ac6ffd700) at pthread_create.c:333
#14 0x00007f6ba993f41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

corkyw10, 2021-02-25:
@jackbart94 could you follow up on this please?

jackbart94, 2021-03-10:
Without more information, it's impossible to debug. Also, it refers to an older version of both CARLA and the Traffic Manager.  Update your CARLA version to latest and if this still happens, please open a new issue with more details.

closed 2021-03-10 by jackbart94