carla-simulator/carla #3533: [Testing] PR #3532, TrafficLight and TrafficSigns can spawn on sublevels

doterop, 2020-11-02:
This issue is intended to track PR #3532 .

Remember that this feature only works on maps created with sub-levelling, maps with the _opt suffix. The other maps shouldn't be affected.

Need to test:
- Maps with and without sub-levelling support are not affected by this change because TL and TS are already created.
- Maps without sub-levelling support continue to spawn TL and TS properly.
- Maps with sub-levelling support spawn TL and TS on the "trafficlights" sublevel. If the sublevel does not exist they will spawn on the persistent level.

To see in which level has been spawned each actor, in the world outliner select level instead of type as a filter.

Since all the maps at the moment already have the TL and TS created, some steps are needed to test that they spawn properly.
In maps without sub-levelling:
- Look for all the TL, TL groups and TS and remove them.
In maps with sub-levelling:
- Remove the sublevel from the Levels window. This will trigger the default behaviour.
- Add a new empty map with a name that NOT contains "trafficlights". This will trigger the default behaviour.
- Rename the previous level with a name that contains "trafficlights". This will trigger the expected behaviour.

XGodina, 2020-11-05:
Tested in Windows

Traffic Light
Every road crossing has a group that synchronizes the traffic light. If you remove the TS and not the references from that, UE4 crashed.  If you remove that reference, it works as expected.

Traffic Sign
The traffic signs don't work, I've tested in some maps and different signals, and not appear again when run the simulation.

XGodina, 2020-11-20:
It's only speed limit signals that don't spawn.  We don't support it yet

closed 2020-11-20 by XGodina