carla-simulator/carla #3536: Add a new Pedestrian, new hold prop

mzheng27, 2020-11-02:
I am trying to add some new carried props for the walkers: walking aid, cane etc. so the walkers can hold them. 

I am a little confused about how the carriable props get grouped with a walker (I didn't find any Socket in the BP_Walker Blueprint). Can someone help me with the steps of adding these carriable props? 

My other question is, can I add a new walker following similar steps in the "Add new vehicle" tutorial (with a new animation in step 4.4): https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/ 

Thank you!

DnDare, 2020-11-03:
We haven't added any sockets to hold items for our pedestrians.
If you want to add a socket, you'll have to locate the SkeletalMesh, select a joint and add a socket.
[image]

We're not planning to add sockets or items to our pedestrians yet. 
-----
About importing pedestrians, new pedestrians must have a specific skeletal mesh, when imported they must be assigned to the SK_Pedestrian_Generan skeleton.
We don't have a full tutorial on character creation, yet, but we have some information published.
Skeleton hierarchy:
https://carla.readthedocs.io/en/latest/tuto_G_control_walker_skeletons/
And this video from 12:30:
https://youtu.be/Uoz2ihDwaWA?t=746
There's a part about importing pedestrians, and setup.

DnDare, 2020-11-05:
@mzheng27   I guess you've seen the answer, if you have more questions about this pipeline, or specific part , just ask

mzheng27, 2020-11-05:
@DnDare Thank you! Could you explain a few more questions? 

Looking at the BP_Walker Blueprint, the function Prop takes in a "Prop" which has a "Socket Name". What is that Socket Name and where the Prop function get it from? 

I think Prop calls the function AttachToComponent to attach a carrying object with the pedestrian. After I import a prop to Unreal Engine, is there a way to add it to the current carrying list (umbrella/trolley/guitar)?

DnDare, 2020-11-05:
@mzheng27  The function "Prop" and the related list (umbrella/trolley/guitar), is outdated. These were build for older models , long ago discarded and I've never seen them operational.  

I think the best option is that you create a socket at the "SK_Pedestrian_Generan"' SkeletalMesh's joint/bone. (It's the generic skeleton we use), else you can add a socket at selected Skeletal Meshes, from WalkerFactory blueprint´s list. (This is the list of spawnable pedestrians)
[image]

Maybe you can still use some older functions, or modify them, but can't tell whether they're still functional. Also IK related nodes are functional/limited to 2wheeled vehicles's pedestrian riders, not for Props. 

 And later add the prop of your choice, at the Blueprint, I wouldn't use BP_walker, because it might affect all pedestrians, I'd add the prop selected to the appropriate pedestrian from WalkerFactory blueprint's list. 
I picked a generic tutorial I found about adding sockets and props:
https://youtu.be/DyPq1-JGMKY?t=359
I have not fully tested, but I think it will work, else contact me

mzheng27, 2020-11-05:
@DnDare Thank you!! I will test this out Friday.

mzheng27, 2020-11-06:
Works great. Prop gets spawned with the walker. Thanks @DnDare

closed 2021-02-25 by corkyw10

YHao29, 2023-04-14:
Hello guys, I have met the same issue.
### Prepares
I use the **docker version of carla 0.9.13**  and I have design a new prop then import to the docker following this [guide.](https://carla.readthedocs.io/en/latest/tuto_A_add_props/) .
After I run the ` ./ImportAssets.sh`, I can see a ` Package01` has been created in the filefolder :
[image]

### Problem
But when I want to spawn it to the world, I can not find it in the blueprint. And I also don't know how to use it like other props (` 'static.prop.bench01'`). 

### Question
Could you teach me that how to use the props I just import to the **carla docker**? thanks for your time

adityadikshit07, 2023-04-15:
Did you find a solution to this ? @YHao29

YHao29, 2023-04-16:
> Did you find a solution to this ? @YHao29

No, I have no idea to deal with it